betaflight/betaflight #13361: At 4.4.3 either motors or ledstrip work

gumileeev, 2024-02-12:
### Describe the bug

I have a stock [2022 Meteor65 ELRS](https://betafpv.com/products/meteor65-brushless-whoop-quadcopter-1s) with an [F4 1S 5A FC (2022)](https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-5a-aio-brushless-flight-controller-elrs-2-4g) flight controller installed on it. I updated the firmware to 4.4.3, after which I decided to add LEDs. If I activate ledstrip in the configuration, on the motors tab I start getting the “The Quad X model need 4 motor resources and the current configuration of the firmware provides 0 usable output(s) for the selected mode. if  using a custom mixer mode you need to define a custom mmix before changing mixer mode. Please check your configuration and add the required motor resources.” error. In this case, the backlight works and is adjusted. As soon as I remove the ledstrip in the settings, the motors tab stops giving an error, I can start the motors in the section and fly around the house. I tried entering "dshot_bitbang = off" through the command line, after that I can start the motors in the motors tab, there are no errors. But if I want to fly around the apartment, I get a CALIBRATING prohibition flag. What's wrong with my setup? I am attaching 2 dumps.
[Meteor65_elrs_4.4.3_dump_ledstrip_on motor_error.txt](https://github.com/betaflight/betaflight/files/14235902/Meteor65_elrs_4.4.3_dump_ledstrip_on.motor_error.txt)
[Meteor65_elrs_4.4.3_dump_no_ledstrip.txt](https://github.com/betaflight/betaflight/files/14235904/Meteor65_elrs_4.4.3_dump_no_ledstrip.txt)

### To Reproduce

Install version 4.4.3 on the specified flight controller and turn on the ledstrip. It turns out you can either fly or shine light bulbs.

### Expected behavior

I really want to fly and shine light bulbs at the same time.

### Support ID

```txt
[Meteor65_elrs_4.4.3_dump_ledstrip_on motor_error.txt](https://github.com/betaflight/betaflight/files/14235902/Meteor65_elrs_4.4.3_dump_ledstrip_on.motor_error.txt)
[Meteor65_elrs_4.4.3_dump_no_ledstrip.txt](https://github.com/betaflight/betaflight/files/14235904/Meteor65_elrs_4.4.3_dump_no_ledstrip.txt)
```

### Flight controller

F4 1S 5A FC (2022) from BetaFPV

### Other components

Everything is stock, just added an RGB LED.

ot0tot, 2024-02-21:
Please submit a Support ID using the "Submit Support ID" button on the CLI tab in Betaflight Configurator, and update your issue with the Support ID.

gumileeev, 2024-02-21:
> Пожалуйста, отправьте идентификатор службы поддержки, используя кнопку "Отправить идентификатор службы поддержки" на вкладке CLI в конфигураторе Betaflight, и обновите вашу проблему с помощью идентификатора службы поддержки.

Id: "eab4fb76-b787-4abc-aba4-6875d4dd037b"

gumileeev, 2024-02-21:
> Please submit a Support ID using the "Submit Support ID" button on the CLI tab in Betaflight Configurator, and update your issue with the Support ID.

Id: "eab4fb76-b787-4abc-aba4-6875d4dd037b"

ot0tot, 2024-02-21:
Which version of the flight controller do you have?

There is a known issue with the V2.1 board where timer based DShot does not work, and you must use bitbanged DShot: https://github.com/betaflight/betaflight/issues/13102

Unfortunately, due to a chip errata with the F4, using bitbanged DShot means other devices on DMA2 (gyro, LED_STRIP) do not get DMA.

In short, if you have version 2.1 of the FC, you cannot use LED_STRIP.

gumileeev, 2024-02-21:
> Which version of the flight controller do you have?

You are absolutely right, my board version is v2.1. It is sad :(

closed 2024-03-30 by github-actions[bot]