betaflight/betaflight #13362: Return to home auto disarmed 50m in air.

11BEX11, 2024-02-12:
### Describe the bug

While i was testing betaflight RC2 the RTH made drone came home but it flew a strange way and than it disarmed more than 50m above home point. the drone crashed on my roofband fall on the street and it burned the battery...i was quick enaugh to put the battery of the drone before it burned..the drone flew in a wery scary way..nose up and down all around and then it tilted and disarmed..i will add the blackbox data and the video when i get home.. 

### To Reproduce

dont know what to make..everithing is set like in the description..the magnetometer was not in use at the time of the flying..

### Expected behavior

please look in the blackbox and write if there was my error or it is something wrong with the code..

### Support ID

```txt
/
```

### Flight controller

fc stack speedybee f405v4

### Other components

 -gps geprc m10 dq 
 -motors aos supernova
-props dalprop 5147 spitfire green
-rx radiomaster elrs rp3
-tx radiomaster ranger 
-googles dji v2 
- dji 03 unit.

### Add any other context about the problem that you think might be relevant here

will add blackbox data and video as i get home

sugaarK, 2024-02-19:
our code is NOT Return To Home..... its Rescue. you are meant to regain control when it comes back to you.. while if you TUNE the rescue parameters properly it can return to home it is not meant to be a RTH system.  you should do testing in a nice big park over grass and once your happy with it use it a RESCUE system like it was intended...

davetheghost, 2024-02-29:
keep an eye on your altitude in telemetry -- was doing some testing and it took off vertically _twice_ -- altitude was 0 when i got it back under control to glance down

11BEX11, 2024-03-01:
i think that is something there.. it was around 50m but it was around 110 in osd...i think that baro from gps and my fc was  somehow an issue...i didnt make another flight till today becouse of the weather and no time..

davetheghost, 2024-03-01:
I should add, i keep a piece of foam over the barometer on my FC to reduce abrupt pressure changes from things like wind, and vortexes caused by the props. After the first fly-away-- I immediately switched off GPS Rescue but kept it armed the entire time, flew it back down to where I was, and disarmed and rearmed to establish a new land location. Tried it again, and that's when it flew away the second time. The moment I noticed it was going to ascend and not stop (can usually tell by the speed of throttle it applies, also surpasses the height you choose, I switched off rescue again, at about 250~300ft. Had to have it fall about 100~150ft with no throttle, landed, and this time I rebooted the FC. After I did that, I was getting what I assume correct Altitude readings and tried a third time, after zipping around and monitoring altitude. 

I guess the saying, 'third time is a charm', reigns true, though that's based on 50/50 odds with each half arising once out of two iterations. Third time is the ultimate result.

ctzsnooze, 2024-04-06:
Please note that not all Barometers are accurate.  Some are very inaccurate.  The real Infineon ones are good, but many are clones and have bad data.  
Need to log them and compare to GPS, or arm, fly up a known height and hover there for a period of time if you're having issues.
Betaflight prefers the Barometer data so if you have serious altitude control issues, usually it's the barometer.
It can sometimes be the GPS losing a lot of satellites if the original 3D fix was weak.  Show PDOP in the goggles.
Please use nightly Betaflight version if you had an issue with Rescue in an older version.
Always test at low altitude over soft grass, and always start only with GPS-only rescue.

closed 2024-04-06 by ctzsnooze