betaflight/betaflight #13365: High RPM Error on 3 motors (with DSHOT600) or 2 motors (with DSHOT300)

drdrechronik, 2024-02-12:
### Describe the bug

[image]
[image]
[TASKS_BTFL_cli_20240212_195201_IFLIGHT_BLITZ_F435.txt](https://github.com/betaflight/betaflight/files/14248682/TASKS_BTFL_cli_20240212_195201_IFLIGHT_BLITZ_F435.txt)

Built a new drone with the iFlight Blitz Mini F435 E55S Stack. With RPM-Filtering turned off and the ESC-protocol set to DSHOT600, the drone flies fine. When turning RPM-Filtering on I noticed only motor 4 had 0% errors, when changing the protocol to DSHOT300 motor 4 and motor 2 worked.

### To Reproduce

I have reproduced this issue with multiple versions of BlueJay 0.16, 0.17 and 0.19 each at 24kHz and 48kHz, for the sake of it also reproduced the issue when i had all UARTS disabled (in an attempt to minimze CPU load, see TASKS of that attempt attached).
[TASKS_NO_UART_BTFL_cli_20240212_200415_IFLIGHT_BLITZ_F435.txt](https://github.com/betaflight/betaflight/files/14249360/TASKS_NO_UART_BTFL_cli_20240212_200415_IFLIGHT_BLITZ_F435.txt)

### Expected behavior

Since control of the ESC via DSHOT seems to be working, I'd also expect bi directional DSHOT with RPM Filtering to work on all 4 motors.

### Support ID

```txt
c6d4ccdd-8ccd-4926-b8c3-40309594d534
```

### Flight controller

iFlight BLITZ Mini ATF435 Flight Controller, bought from iFlight

### Other components

iFlight BLITZ E55S 4-IN-1 ESC

### How are the different components wired up (including port information)

ESC and FC came as a stack bundle, plugs attached as recommended by iFlight.

haslinghuis, 2024-02-12:
Enable `set dshot_bitbang = ON` should fix it.

See also: https://github.com/betaflight/config/pull/320

drdrechronik, 2024-02-12:
> Enable `set dshot_bitbang = ON` should fix it.
> 
> See also: [betaflight/config#320](https://github.com/betaflight/config/pull/320)

Thank you so much!

closed 2024-02-12 by haslinghuis