betaflight/betaflight #13368: CRAZYBEEF4SX1280 VERY SLOW OSD UPDATE BF 4.5RC2

ivan-ef, 2024-02-13:
### Describe the bug

FC: CRAZYBEEF4SX1280 (Happymodel X12 aio SPI-ELRS v2.1)
Flashing from BF configurator 10.10.0-RC2
Apply default settings.
Very long OSD update time (sometimes even few seconds).

### To Reproduce

Flash to BF4.5-RC2
Apply default settings.
Add few OSD elements like compass, virtual horizon  manually.

### Expected behavior

working correct in BF 4.4.3

### Support ID

```txt
34f6c445-ceaa-4f1d-82ef-40f3ff5e9d9b
```

### Flight controller

CRAZYBEEF4SX1280

ivan-ef, 2024-02-15:
I tried my other x12 aio FC which has external uart receiver and it works ok.

haslinghuis, 2024-02-15:
Blackbox and OSD share same SPI bus. 

Try 
- disable blackbox.
- enable bitbang
- disable led_strip

ivan-ef, 2024-02-16:
> Blackbox and OSD share same SPI bus.
> 
> Try
> 
> * disable blackbox.
> * enable bitbang
> * disable led_strip

Tried but no luck.

SPI receiver is affects OSD update speed. I disabled SPI receiver and OSD updates much faster.

haslinghuis, 2024-02-16:
Lower the PID loop rate to 2Khz:

https://betaflight.com/docs/development/manufacturer/manufacturer-design-guidelines#41-rated-looptime-and-performance

ivan-ef, 2024-02-17:
> Lower the PID loop rate to 2Khz:
> 
> https://betaflight.com/docs/development/manufacturer/manufacturer-design-guidelines#41-rated-looptime-and-performance

This is it. Thanks

closed 2024-02-17 by ivan-ef