betaflight/betaflight #13374: Do not go into failsafe when roll/pitch/yaw/throttle are sent via MSP (based on MSP_OVERRIDE mask) and no connection to RC.

yrik, 2024-02-17:
### Is your feature request related to a problem? Please describe

I'm making a project with an autonomous drone, it needs to be agile, that's why I use Betaflight.
I have a companion computer that sends control (roll/pitch/yaw/throttle) signals to the flight controller using the MSP_OVERRIDE mechanism. It works fine, however when I disconnect remote controller then the drone goes into failsafe mode, even though the correct controls signal are still sent via MSP.

### Describe the solution you'd like

a) the check for control signals turns on failsafe mode should consider MSP signals as well, not only RC.
b) an option to disable failsafe?

### Describe alternatives you've considered

I've tried to set longer stage 1 in failsafe settings, but looks like when it goes directly into stage 2 when rc is not connected. I've tried to modify pulse min value in betaflight configurator to disable failsafe, but it's not possible to set it smaller than 750.

yrik, 2024-02-17:
Looks like the flow is stopping here 

https://github.com/betaflight/betaflight/blob/0daed9f7dc42612ac862394ae41977f57c8fedc2/src/main/rx/rx.c#L765

without checking that data has arrived in https://github.com/betaflight/betaflight/blob/0daed9f7dc42612ac862394ae41977f57c8fedc2/src/main/rx/rx.c#L780

haslinghuis, 2024-02-17:
@yrik Sounds like a bug instead of feature request. Can you give more information about the setup (understand using a receiver and an offboard SBC).

- which receiver is installed (protocol, port setting)
- how is MSP setup for the companion board sending data (protocol, port setting)

Option a seems the way to go (as we need to ensure valid data incoming from MSP if RX is in failsafe).

yrik, 2024-02-17:
@haslinghuis Thank you

> - which receiver is installed (protocol, port setting)

Receiver is connected via SERIAL protocol, ELRS receiver (crsf is selected in betaflight configurator)

>- how is MSP setup for the companion board sending data (protocol, port setting)

I have tried it with two different setups, one is connected via USB/MSP, and another is connected via SERIAL port.

haslinghuis, 2024-02-17:
@yrik can you test the change. Assume `MSP_OVERRIDE` is used using an AUX switch to activate. Disabling the AUX switch would go into fail-safe if the primary RX is not sending signal and / or `BOXFAILSAFE` is activated.

yrik, 2024-02-18:
Thank you for the fix, I'm trying to test it but I'm stuck a bit with configuring a custom version of betaflight.

I've built it with the command 
```
make STM32F405 EXTRA_FLAGS="-D'BUILD_KEY=18d578761cda9996c31cd574cb929eee' -D'RELEASE_NAME=4.5.0' -D'BOARD_NAME=AOCODAF405V2MPU6500' -D'MANUFACTURER_ID=SJET' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6500 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_BARO_DPS310 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_MPU6500 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"
```
and flashed with betaflight configurator. I've also loaded backup cli config after flashing (there were several errors indicated). But after that, the accelerometer and gyro are not available. If I flash it again with a standard 4.4 version it suggests to apply custom defaults and accel/gyro are available.

[image]
[image]
[image]
[image]

haslinghuis, 2024-02-18:
For 4.5 use  `make AOCODAF405V2MPU6500 EXTRA_FLAGS=" ... "`
Only import relevant config items as things have changed.

yrik, 2024-02-18:
Thanks, I was able to build, flash and run the modified version of betaflight. But unfortunately, it seems that it did not solve the original issue. It still shows RXLOSS on the OSD and goes into failsafe when I turn off RC (with MSP_OVERRIDE commands running)

haslinghuis, 2024-02-18:
Assume using pinio box 50 (BOXMSPOVERRIDE) to activate MSP_OVERRIDE  and having set msp_override_channels_mask?

Do you have a diff of the configuration?

yrik, 2024-02-18:
Yes, exactly. 
My testing process is following:
- I arm
 - then switch to MSP mode using switch on remote controller bind to AUX3 (MSP commands are sent every 30-40ms). 
 - I see that speed of motors is changing and then
 - and then turn off remote controller.
 - I see that drone instantly goes into RXLOSS on OSD and then turns off after end of stage 1. 

Here is the diff and screenshots

[BTFL_cli_20240218_223001_AOCODAF405V2MPU6500.txt](https://github.com/betaflight/betaflight/files/14324519/BTFL_cli_20240218_223001_AOCODAF405V2MPU6500.txt)

[image]
[image]

haslinghuis, 2024-02-18:
Ah, switching off the TX will alter the AUX3 setting (disabling MSP OVERRIDE)
Can test with setting AUX3 on failsafe tab setting AUX3 to hold

yrik, 2024-02-18:
I've set failsafe stage 1 settings to hold, so was expecting that it would keep the high signal even if RC is off. I can test tomorrow with a fixed value as well.

haslinghuis, 2024-02-18:
Using same condition as here in my PR:

https://github.com/betaflight/betaflight/blob/0daed9f7dc42612ac862394ae41977f57c8fedc2/src/main/rx/msp_override.c#L31-L43

Curious about companion computer[s] used.

yrik, 2024-02-18:
The `rxMspOverrideReadRawRc ` part definitely works, because I see that roll/pitch/yaw/roll values are the same that are sent via MSP. I'm using raspberry pi 4 in this case, it's connected via USB to flight controller. 

I will try tomorrow for the start just to remove all conditions inside of `calculateRxChannelsAndUpdateFailsafe`  for the test and to see if it helps. But I am afraid that there could be some extra logic we are not taking into account.

haslinghuis, 2024-02-18:
Agree. We would also need fail-safe for MSP override.

yrik, 2024-02-19:
@haslinghuis 

This change solves the issue according to my tests with hardware. 
Basically, it overrides variables on an earlier step so that MSP controls check is considered as well.

https://github.com/betaflight/betaflight/pull/13380/files

haslinghuis, 2024-02-19:
Does fail-safe still work if MSP control is stalled?

yrik, 2024-02-19:
let me test it

yrik, 2024-02-19:
> Does fail-safe still work if MSP control is stalled?

Nope :( 
I was expecting `detectAndApplySignalLossBehaviour` to handle it because signal would be 0, but looks like in msp code it just takes the last set value.

yrik, 2024-02-19:
Do you mean the current behavior in master or in this PR is correct? 

With the implementation in this PR it would enable autonomous flight controlled by companion computer, otherwise it's not possible. 

Also according to the failsafe documentation it should trigger when there is no roll/pitch/yaw/throttle signals. Or  a when all channels are missing. In this case the roll/pitch/yaw/throttle are sent continuously via MSP. 

The functionality of switching MPS control can be turned on and off with the RC link if it's present. And yes, local MSP most likely not break, but it's also possible to handle this case.

yrik, 2024-02-19:
I've updated the PR to handle failsafe in MSP_OVERRIDE mode. It's more uniform now with other providers of controls like RC. It will go into failsafe when there are no roll/pitch/yaw/throttle controls received on any of the channels.

https://github.com/betaflight/betaflight/pull/13380/files#diff-fd7378e061996b8807886db1284826871335e87582d4680167f2cec73d6eb3f0R571

antonpolishko, 2024-02-28:
This is exactly the blocker raised amongst our group of AI enthusiasts playing with betaflight. Following the thread

PR #13380 (2024-03-05): Do not go into failsafe when the correct roll/pitch/yaw/throttle are sent via MSP_OVERRIDE
Fixes #13374

Do not go into failsafe when the correct roll/pitch/yaw/throttle is sent via MSP_OVERRIDE.
Make sure to go into failsafe when there are no controls via RC link and via MSP_OVERRIDE.