betaflight/betaflight #13382: Possible bug with mamba H743 FC and M10 GPS on BF 4.5 RC 1 & 2

Atarix01, 2024-02-19:
### Describe the bug

Hello, I have 3 drones with mamba H743 and Foxeer M10Q GPS. 

1 week ago, I updated one of these drones, at first glance everything is fine but after two flights, the FC LEDs ( the blue and orange ) flash as if it were restarting, however the drone does not shut down in flight, it continues to fly normally. But when you activate RTH, the drone falls after 1 second.

I changed the GPS with that of my other functional drones, same bug. I downgraded the firmware to 4.4.3 the bug persists. I bought a new FC, flashed with BF 4.5.RC2, I had the same bug. 
When I click on "save and reboot" the FC crashes (the LEDs stay on and nothing happens).
I changed the GPS with old bn220, obviously everything works normally . 
I'm not sure of the source of the bug but I don't see any other possible solution.

( Sorry for my poor English, I am french lol )

### To Reproduce

Flash betaflight 4.5 on a Mamba H743 FC and put a M10 GPS ( Foxeer in my case ) and fly 2/3 li-po. I you just wait without arm/fly it looks like it doesn't work/bug.

### Expected behavior

The LEDs of the FC start to blink like if the FC is rebooting every second. When you click on "save and reboot" in betaflight configurator, the FC crash, and the LEDs stay ON, then you have to power of to unlock the FC.

### Support ID

```txt
00a28c66-52a1-40a0-9ec3-1fb9489eb68e
```

### Flight controller

Mamba H743 ( target MAMBA H743_2022B

### Other components

ESC.    T-motor Velox V50
RX.    ELRS nano RX
GPS.    Foxeer M10Q 250
LEDs.     mamba/flywoo
Buzzer.     Vifly Finder mini
Motors.    Xnova Smooth line
VTX.    Walksnail AVATAR 

### How are the different components wired up (including port information)

RX UART1
WIFI UART4
VTX UART3
GPS UART7

haslinghuis, 2024-02-19:
Please try with latest configurator from https://github.com/betaflight/betaflight-configurator-nightlies/releases and use latest development firmware.

Atarix01, 2024-02-19:
Ok, I will try that. I am going to test the drone tomorrow.  
I'll tell you what happens 👌

haslinghuis, 2024-02-20:
So bug happens only with GPS attached?

Support ID is from 4.4.3. Please leave out OSD_SD from the build if using Walksnail. There are issues with using W25N01G flash chip.

Atarix01, 2024-02-21:
Okay 👍. Yes the bug appears only when the GPS is plugged. 
I'll try to re-flash my FC without OSD_SD and try to fly with that.

Atarix01, 2024-02-24:
So I have tested with BF 4.5 RC2, without OSD_SD and with a new M10 GPS from an other brand ( GEPRC ) and the bug is still here 😭

Atarix01, 2024-02-24:
I'll try to plug the GPS on an other UART to see if it works

Atarix01, 2024-02-24:
The bug is still here even if I change the UART. I haven't anymore ideas to solve the problem, I think I'll buy a new one... Again

haslinghuis, 2024-02-24:
Why is MAMBA H743 not listed on the diatone site? The flash chip could have a [huge] negative impact too.

Atarix01, 2024-02-25:
https://www.diatone.us/products/mb-mk4-h743-v2-fc It's this one. The flash chip could have an impact, so it's an hardware problem. Can I solve it without buying new FC or solder a new flash chip : ) ?

closed 2024-04-04 by github-actions[bot]