betaflight/betaflight #13389: GPS 3D Fix Losing Connection | Losing Satellite Count

spatzengr, 2024-02-25:
### Describe the bug

In BF 4.5, the satellite lock / 3D Fix is dropping out.  This does not happen in BF 4.4.3 with the same equipment.  See https://app.screencast.com/H3Ehs43XuFGH6 for a demonstration of the issue.

### To Reproduce

SpeedyBeeF7V3 with M81-5883 GPS.

### Expected behavior

Expect GPS to keep 3D Fix lock.

### Support ID

```txt
023633fd-2724-46f5-9ee6-13d82c66c706
```

### Flight controller

SpeedyBeeF7V3

ctzsnooze, 2024-02-25:
Both logs show a failure to receive valid data packets and hence a failure to connect successfully.

The baud rate appears to negotiate normally, but no valid data packets are received.

Please try a different GPS module on the same build.  

When interpreting the timing debug, the states are encoded as three numbers; eg 502 means main state 5, sub-state 2.

The log shows the unit cycling through repetitive connection attempts with no valid data received at any point, eventually leading to ack failures and an attempt to reconnect 

```
typedef enum {
    UBLOX_DETECT_UNIT,      //  0
    UBLOX_SLOW_NAV_RATE,    //  1. 
    UBLOX_MSG_DISABLE_NMEA, //  2. Disable NMEA, config message
    UBLOX_MSG_VGS,          //  3. VGS: Course over ground and Ground speed
    UBLOX_MSG_GSV,          //  4. GSV: GNSS Satellites in View
    UBLOX_MSG_GLL,          //  5. GLL: Latitude and longitude, with time of position fix and status
    UBLOX_MSG_GGA,          //  6. GGA: Global positioning system fix data
    UBLOX_MSG_GSA,          //  7. GSA: GNSS DOP and Active Satellites
    UBLOX_MSG_RMC,          //  8. RMC: Recommended Minimum data
    UBLOX_ACQUIRE_MODEL,    //  9
//    UBLOX_CFG_ANA,          //  . ANA: if M10, enable autonomous mode : temporarily disabled.
    UBLOX_SET_SBAS,         // 10. Sets SBAS
    UBLOX_SET_PMS,          // 11. Sets Power Mode
    UBLOX_MSG_NAV_PVT,      // 12. set NAV-PVT rate
    UBLOX_MSG_SOL,          // 13. set SOL MSG rate
    UBLOX_MSG_POSLLH,       // 14. set POSLLH MSG rate
    UBLOX_MSG_STATUS,       // 15: set STATUS MSG rate
    UBLOX_MSG_VELNED,       // 16. set VELNED MSG rate
    UBLOX_MSG_DOP,          // 17. MSG_NAV_DOP
    UBLOX_SAT_INFO,         // 18. MSG_NAV_SAT message
    UBLOX_SET_NAV_RATE,     // 19. set to user requested GPS sample rate
    UBLOX_MSG_CFG_GNSS,     // 20. For not SBAS or GALILEO
    UBLOX_CONFIG_COMPLETE   // 21. Config finished, start receiving data
} ubloxStatePosition_e;

typedef enum {
    GPS_STATE_UNKNOWN = 0,
    GPS_STATE_DETECT_BAUD,      // 1
    GPS_STATE_INITIALIZED,      // 2
    GPS_STATE_CHANGE_BAUD,      // 3
    GPS_STATE_CONFIGURE,        // 4
    GPS_STATE_RECEIVING_DATA,   // 5
    GPS_STATE_PROCESS_DATA,     // 6
    GPS_STATE_LOST_COMMUNICATION, //7
    GPS_STATE_COUNT             //8
} gpsState_e;

typedef enum {
    UBLOX_ACK_IDLE = 0,
    UBLOX_ACK_WAITING,      // 1
    UBLOX_ACK_GOT_ACK,      // 2
    UBLOX_ACK_GOT_NACK      // 3
} ubloxAckState_e;
```

ctzsnooze, 2024-02-25:
Not all GPS units that work with 4.4 will work with 4.5.

ctzsnooze, 2024-04-06:
Closing, but re-open if you have an update.

closed 2024-04-06 by ctzsnooze