betaflight/betaflight #13391: Assisted Flip Over After Crash (Assisted Turtle)

kevinsumner, 2024-02-25:
### Is your feature request related to a problem? Please describe

Some pilots, especially those of us running tiny whoops, find themselves having difficulty when using Flip Over After Crash. For two particular examples, overshooting upright and ending up back upside down, and some crafts (again, usually tiny whoops) don't have enough power on defaults to flip the craft over.

### Describe the solution you'd like

A new (and optional?) feature for FOAC to help automate the flip, where Betaflight only takes an initial input from the pilot as to which way to flip, then handles the flip autonomously. As a naive, mock process, BF could ramp the relevant motors until significant movement is detected then hold until the gyro detects the craft has passed some angle beyond vertical.

### Describe alternatives you've considered

Adding the CLI tuning options for FOAC as UI elements to the BF Configurator somewhere and improving documentation around those FOAC options. (`crashflip_motor_percent` and `crashflip_expo` are the only two afaik, although the other `crash_` options might be relevant). Also, maybe add Preset to tune those better for Tiny Whoops?

### Other information

Quicksilver pilots call this "Assisted Turtle". It might be useful to take inspiration from the QS implementation.

FPVSkittles, 2024-02-26:
I second this request. When I attempt to crash flip with a whoop in betaflight I am lucky if I get it in the first 3 tries, somtimes you flip too little, then too much, it's hard to get just right. When using quicksilver the assisted turtle mode flips me over 9/10 times on the first try. It would be great if this was just another mode avaiable to set a aux channel to and use. You could even traditional crash flip on one switch and assisted on another. I hope we can add this.

ChrisRosser, 2024-03-19:
Nice idea. One approach could be to limit the max rotation rate to something pretty low like 100 deg/s when in FOAC mode. This should greatly reduce the chance of overshooting.

jawz101, 2024-05-22:
Because my mind is slow, sometimes I forget which direction is the direction I really want to flip and my quad may favor one direction more once I try nudging it in both directions (e.g. a twig is on the side that initially looks ideal.)

I think it would make sense to have some sort of delay before performing the assist.  Like I wiggle my sticks in either direction is probably around a 200-400ms movement but if I'm commited to a side - because I've throttled up higher or am holding the stick in that direction for more than 400ms- then kick in the assist.

kevinsumner, 2026-01-18:
Yes, I believe this is resolved with 2025.12.1, although I haven't tried it yet. Marking this as closed/completed.

closed 2026-01-18 by kevinsumner