betaflight/betaflight #13393: 4.5-RC2 does not support SPI receiver, can not match frequency?

Tang346, 2024-02-26:
### Describe the bug

I recently made a flight control of a built-in SPI receiver, the main control chip is AT32F435CGU7, the wireless chip uses SX1280, and the SPI+ELRS protocol is used for communication. I tried to use the ELRS remote control for frequency alignment, and the oscilloscope couldn't capture the SPI SCK timing.

### To Reproduce

My target configuration for RX SPI is as follows:
```c
/* ======== SPI ======== */
#define USE_SPI
#define USE_SPI_DEVICE_1
#define SPI1_SCK_PIN            PA5
#define SPI1_MISO_PIN           PA6
#define SPI1_MOSI_PIN           PA7

/* ======== RX ======== */
#define USE_RX_SPI
#define RX_SPI_INSTANCE        SPI1
#define RX_SPI_LED_INVERTED

#define USE_RX_EXPRESSLRS
#define USE_RX_EXPRESSLRS_TELEMETRY
#define USE_RX_SX1280
#define RX_CHANNELS_AETR
#define RX_SPI_BIND_PIN        PH2
#define RX_NSS_PIN             PA4
#define RX_SPI_LED_PIN         PC14
#define RX_SPI_EXTI_PIN        PB9
#define RX_EXPRESSLRS_SPI_RESET_PIN      PB4
#define RX_EXPRESSLRS_SPI_BUSY_PIN       PH3
#define RX_EXPRESSLRS_TIMER_INSTANCE     TMR5

#define DEFAULT_RX_FEATURE     FEATURE_RX_SPI
#define RX_SPI_DEFAULT_PROTOCOL RX_SPI_EXPRESSLRS
```

### Expected behavior

If you do not use RX_SPI and use an external serial receiver, you can use the frequency normally.

### Support ID

```txt
/
```

### Flight controller

The main control chip uses AT32F435CGU7, and the hardware is fine

### Other components

.

### How are the different components wired up (including port information)

.

### Add any other context about the problem that you think might be relevant here

.

haslinghuis, 2024-02-26:
New designs should not include a SPI receiver but use U[S]ART instead.

Tang346, 2024-02-27:
> New designs should not include a SPI receiver but use U[S]ART instead.

Thanks, but I wonder why SPI is not included, bf4.4 is

Tang346, 2024-02-28:
> *** There are no SPI Rx solutions that are strongly recommended for future development due to challenges in resource allocation and scheduler inconsistency that consistently emerge with SPI Rx designs. Additionally, there are no RC ecosystems that are actively developing a supported SPI Rx solution (ExpressLRS 3.0 and later do not support SPI; FrSky does not support SPI Rx over any protocol, and other SPI Rx solutions have been fully deprecated).
> 
> Former STM32 targets will still support ELRS SPI v3 on BF 4.5 but won't update to v4.
Thanks !

closed 2024-03-19 by haslinghuis