betaflight/betaflight #13396: ESC Not Connecting in BLHELI 32 in 4.5

Slatsks, 2024-02-26:
### Describe the bug

TMotor F7 Pro, Hobbywing 65Amp 4 in 1.  ESC Not recognized in BLHeli 32

### To Reproduce

Connects fine in other Betaflight Versions

### Expected behavior

I relashed back to the older Betaflight Firmware and it connects no problem.  This is a new build.  

### Support ID

```txt
standard set up
```

### Flight controller

TMotor F7 Pro

### Other components

Hobbywing 65Amp 4 in 1

### How are the different components wired up (including port information)

Plug

### Add any other context about the problem that you think might be relevant here

The ESC is not reading with BF 4.5 and BL Heli 32.  I relashed with older BF Versions and it connected no problem.

haslinghuis, 2024-02-26:
Fixed in [master.](https://github.com/betaflight/betaflight/pull/13364)

matt-FPV, 2024-02-27:
I have the same problem with the FC Diatone Mamba Mini Mk4 (Target: MAMBA F722_2022A). Tried with .dev version, also not working. Works fine with BF 4.4 versions.

jaybeefr, 2024-03-01:
hello, i have a hobbywing xrotor f7 stack with 65a esc and I have the same problem. blheli_32 doesn't recognise my esc. I can't flash them, nor do anything else than turn the motors (wich work)
is there a solution ? or do I have to switch to 4.4 ?

jaybeefr, 2024-03-01:
sorry, just understood what you meant in master. I'm new to the hobby and english is not my first language, just trying my best :)
thanks it does work.

haslinghuis, 2024-03-19:
Fixed in RC3

closed 2024-03-19 by haslinghuis