betaflight/betaflight #13399: the final flight time and final distance are not displayed in the dashboard

lyhnsdgtdh, 2024-02-27:
### Describe the bug

the final flight time and final distance are not displayed in the dashboard, I changed the firmware from 4.4.3 to 4.5.0, it didn’t show and still doesn’t show, the last item also shows 0, and when in the arm the satellites are caught, the distance traveled is counted.
[image]
[image]
[image]

### To Reproduce

I need to get in dashboard how long the flight lasted and what distance was covered

### Expected behavior

I need to get in dashboard how long the flight lasted and what distance was covered

### Support ID

```txt
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# support id
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# Submit Support Data
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# Submit Support Data
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# status
MCU F722 Clock=216MHz, Vref=3.25V, Core temp=45degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3604, max available: 16384
Devices detected: SPI:1, I2C:2
Gyros detected: gyro 2 locked dma
GYRO=BMI270, ACC=BMI270, BARO=BMP280, MAG=QMC5883
OSD: MAX7456 (30 x 13)
BUILD KEY: 88e93b0da415604e312386743e83c859 (4.5.0-RC2)
System Uptime: 169 seconds, Current Time: 2024-02-27T16:27:34.649+00:00
CPU:25%, cycle time: 315, GYRO rate: 3174, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 6
SD card: Manufacturer 0xc8, 491520kB, 05/2165, v12.8, 'APPSD'
Filesystem: Ready
GPS: connected, UART6 57600 (set to 57600), configured, version = M9
Arming disable flags: RXLOSS BOXFAILSAFE THROTTLE CLI MSP RESCUE_SW DSHOT_TELEM
```

### Flight controller

speedybeeF7V3

### Other components

[image]

### How are the different components wired up (including port information)

[image]

lyhnsdgtdh, 2024-03-19:
Привет, нету вариантов?

haslinghuis, 2024-03-19:
`GPS: connected, UART6 57600 (set to 57600), configured, version = M9`

What GPS is being used? Can you try with a M10 module?

lyhnsdgtdh, 2024-03-19:
Доброго времени, спасибо что откликнулись! Использую Beitian BK-880Q UBX-M9140, попробовать модуль М10 нет возможности, нужно приобретать дополнительно, а какой бы вы порекомендовали на всякий случай? 
[image]
BK-880Q UBX-M9140

lyhnsdgtdh, 2024-03-19:
такой, здесь характеристики:  https://aliexpress.ru/item/1005002599384876.html?sku_id=12000030828692105&spm=a2g2w.productlist.search_results.0.3e0869a2iD0sVA

haslinghuis, 2024-03-19:
- In the aliexpress link I see M8030-KT in the product info - which is M8. Found BK-880Q UBX-M9140 on another page:

[image]

- Did it work with 4.4.3 ?
- Please submit a support ID. You can find the button in the CLI tab.

lyhnsdgtdh, 2024-03-19:
да, извиняюсь, ссылку неверную дал. у меня M9, BK-880Q UBX-M9140.
на версии 4.4.3. не обращал внимания, сейчас не могу сказать.

lyhnsdgtdh, 2024-03-19:
# ###

# # Problem description

# # the final flight time and final distance are not displayed in the dashboard

# ###

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Feb 25 2024 / 16:41:21 (f1cbd83f0) MSP API: 1.46
# config rev: 5ee9260
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V3

# status
MCU F722 Clock=216MHz, Vref=3.25V, Core temp=51degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3604, max available: 16384
Devices detected: SPI:1, I2C:2
Gyros detected: gyro 2 locked dma
GYRO=BMI270, ACC=BMI270, BARO=BMP280, MAG=QMC5883
OSD: MAX7456 (30 x 13)
BUILD KEY: 88e93b0da415604e312386743e83c859 (4.5.0-RC2)
System Uptime: 388 seconds, Current Time: 2024-03-19T16:56:03.224+00:00
CPU:25%, cycle time: 315, GYRO rate: 3174, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 6
SD card: Manufacturer 0xc8, 491520kB, 05/2165, v12.8, 'APPSD'
Filesystem: Ready
GPS: connected, UART6 57600 (set to 57600), configured, version = M9
Arming disable flags: RXLOSS BOXFAILSAFE THROTTLE CLI MSP RESCUE_SW DSHOT_TELEM

# flash_info
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# dump master

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Feb 25 2024 / 16:41:21 (f1cbd83f0) MSP API: 1.46
# config rev: 5ee9260

# start the command batch
batch start

board_name SPEEDYBEEF7V3
manufacturer_id SPBE

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 A15
resource MOTOR 2 B03
resource MOTOR 3 B04
resource MOTOR 4 B06
resource MOTOR 5 B07
resource MOTOR 6 B05
resource MOTOR 7 B00
resource MOTOR 8 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
[... 976 lines cut ...]

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

*** Support data submitted *** 
Id: "38f16b25-c906-44a0-b0bd-98a0b90832b6"

# copy ID and provide to the betaflight team

lyhnsdgtdh, 2024-03-19:
Вы имели ввиду это? я правильно сделал?

haslinghuis, 2024-03-19:
Please try 4.4.3 - so we know if the GPS module works with that firmware version.

lyhnsdgtdh, 2024-03-19:
хорошо я попробую, но быстро сообщить вам не получится, время ограничено. возможно в течение недели или двух. Подскажите как скопировать настройки чтобы после перепрошивки на 4.3.3. не настраивать дрон по новой на всех вкладках?

lyhnsdgtdh, 2024-03-19:
неточный перевод. 
Okay, I’m trying, but I won’t be able to tell you quickly, time is limited. perhaps within a week or two. Tell me how to copy the settings after flashing to 4.3.3. so as not to configure the drone again on all tabs?

haslinghuis, 2024-03-19:
Use `save backup` in preset tabs for the 4.5 configuration.
Flash 4.4.3 - do some basic setup and test.
Flash back to 4.5 and use `load backup` in presets tabs

Not sure if the issue is related to M9, as our code is optimized for M10 now.

lyhnsdgtdh, 2024-03-19:
да я знакомился с вашими трудами в области оптимизации работы gps в 4.5.0 поэтому и перешел на нее. буду проверять, спасибо за ваш труд!

closed 2024-04-27 by github-actions[bot]