betaflight/betaflight #13402: 4.5-RC1/4.5-RC2 MAG Is Not Working

KenImhof, 2024-02-28:
### Describe the bug

When MAG is enabled, the YAW position shows the same YAW position regardless of the position of the QUAD.  When rotating the QUAD yaw position, the setup window shows the quad moving, but springs back to the original position when the rotation stops.  After performing the Calibrate Magnetometer the mag_calibration does not update and continues to show values 0,0,0.  This behavior occurs in both 4.5-RC1 and 4.5-RC2.  This problem does not exist in version 4.4.3 or in iNav 7.01.

### To Reproduce

With RC1 or RC2, enable MAG and rotate the QUAD and verify the yaw rotation from the configurator setup window.

### Expected behavior

Configuration setup window should show the quads proper yaw position with position value around 0-DEG when the front is position to the north.

### Support ID

```txt
*** Support data submitted *** 
Id: "477eebf0-eae8-4a0e-8dba-9f240cae3293
```

### Flight controller

JHEMCU GF30F722-ICM With Target=JHEF7DUAL

### Other components

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=51degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 2881, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 2 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310, MAG=HMC5883
BUILD KEY: 15bc634bffde45a56adcc3b6f7886e76 (4.5.0-RC2)
System Uptime: 29 seconds, Current Time: 2024-02-28T03:32:33.741+00:00
CPU:31%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 1214 * 0.01V (3S battery - OK)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: connected, UART1 57600 (set to 57600), configured, version = M8
Arming disable flags: RXLOSS CLI MSP

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Feb 26 2024 / 18:23:48 (f1cbd83f0) MSP API: 1.46
# config rev: 48101f1
# board: manufacturer_id: JHEF, board_name: JHEF7DUAL

haslinghuis, 2024-02-28:
Duplicate of #13338

ctzsnooze, 2024-04-06:
Please read the wiki https://betaflight.com/docs/wiki/guides/current/Magnetometer
Please note recent merging of https://github.com/betaflight/betaflight/pull/13487, and test the latest nightly build of Betaflight.

closed 2024-04-06 by ctzsnooze