betaflight/betaflight #13413: OSD stucks on main screan when use SD card

CyberDex, 2024-03-03:
### Describe the bug

When using the SD card on speedebee f405 v3 OSD stuck on the main screen and never change

### To Reproduce

- Install the SD card
- observe OSD never change from the main screen

### Expected behavior

- Install the SD card
-OSD should show main screen and go to to the fly state

### Support ID

```txt
1ef02243-c0a6-49ad-94a7-010ef9e3a4fa
```

### Flight controller

Speedebee f405 v3

CyberDex, 2024-03-03:
Other person facing this issue: https://www.reddit.com/r/fpv/comments/18yl5pk/osd_issue_afte_betaflight_440/

haslinghuis, 2024-03-03:
This is a known issue:

See https://github.com/betaflight/betaflight/issues?q=is%3Aissue+speedybeef405v3+is%3Aclosed

closed 2024-03-03 by haslinghuis