betaflight/betaflight #13415: Add OSD warning for vTX "device ready: no"

joshuabardwell, 2024-03-03:
### Is your feature request related to a problem? Please describe

Currently, Betaflight has no OSD warning if the SmartAudio (or other protocol) connection to the vTX is down. As a result, users may see TX power, band, and channel in the OSD when those values are not correct. This can result in the user flying away and losing video unexpectedly (e.g. the vTX is at 25 mW or in pit mode, or the vTX is on an adjacent channel). 

### Describe the solution you'd like

Add a new Warning message in the OSD for when the FC has "Device Ready: No" status to the vTX. Currently, the configurator shows this status, so it should be possible for the FC to evaluate this status. 

In the vTX settings OSD menu, add a message on screen to clealry indicate "VTX OFFLINE" or something like that, so the user can know that the changes they are making in the FC config will not be received by the vTX.

### Describe alternatives you've considered

None

### Other information

None

joshuabardwell, 2024-03-03:
This warning status should not prevent arming, since users may choose to manually set the vTX settings via button press, or users may fly LOS. But it shouldn't be possible for them to take off without knowing that their assumptions about the vTX may be wrong.

bigspeedfpv, 2024-03-04:
this seems pretty trivial to implement - I could probably cook something up tomorrow (assuming nobody more familiar with betaflight beats me to it)

to be clear, I haven't used the OSD menus for VTX control aside from HDZero in a while - doesn't the SmartAudio menu already show something different (statx or something?) when your VTX isn't connected?

also, would it be worth adding as an LED warning of some sort? It could be useful to have as a way to troubleshoot no video (VTX connection issue vs pit mode for example) but it would also add to the clutter of LED codes that are already there

closed 2026-09-26 by blckmn