betaflight/betaflight #13416: MSP_OVERRIDE commands not being executed

defrag-bambino, 2024-03-04:
### Describe the bug

When enabling MSP_OVERRIDE mode after arming, the motors stop instead of executing the commands send via MSP. But the commands are received correctly and displayed in the RAW_RC message. However, the MOTORS message shows that the motors are indeed not moving (despite the seemingly correct RC commands). 
Also discussed [here](https://github.com/christianrauch/msp/issues/62) in the MSP C++ library.

### To Reproduce

4.5-RC2 Betaflight
Christian Rauch's C++ MSP library, speficifally [this](https://github.com/christianrauch/msp/pull/60) PR (regular main branch does not work).
`set msp_override_channels_mask = 245760`
Use any free channel to enable MSP_OVERRIDE mode, after arming

### Expected behavior

Once switching into MSP_OVERRIDE mode, the RC commands send via MSP should be executed.

### Support ID

```txt
9102aab7-5c6b-42ce-af7d-5d7abf2bc251
```

### Flight controller

SpeedyBee F405 V3 + ESC Stack

### Other components

just motors

### How are the different components wired up (including port information)

I connect the FC to my PC running the MSP stuff via USB-C.

ledvinap, 2024-03-04:
245760 seems to be bit-reversed - AFAIK msp_override_channels_mask maps channels to bit position -  bit0 <-> ROLL, bit1 <-> PITCH, ... 
Maybe you want `msp_override_channels_mask = 15` ?

defrag-bambino, 2024-03-05:
this indeed was the problem. :facepalm: 
Thanks!
Maybe there should be some more clear documentation about this though!

closed 2024-03-05 by defrag-bambino