betaflight/betaflight #13418: Gyros detected: gyro 1 locked dma shared

fedezcason, 2024-03-04:
### Describe the bug

Gyros detected: gyro 1 locked dma shared

### To Reproduce

Gyros detected: gyro 1 locked dma shared

### Expected behavior

gyro fix

### Support ID

```txt
32a1f394-fe55-4800-a782-2b2e13724caf
```

### Flight controller

HGLRCF722MINI

haslinghuis, 2024-03-04:
GYRO 1 is shared due to the following:

```
set baro_bustype = SPI
set baro_spi_device = 1
```

The [target config](https://github.com/betaflight/config/blob/master/configs/HGLRCF722MINI/config.h) uses I2C for baro

fedezcason, 2024-03-04:
Nothing happened.
still Gyros detected: gyro 1 locked dma shared

SteveCEvans, 2024-03-04:
What’s the supposed bug?

haslinghuis, 2024-03-04:
Above shows what is wrong with the configuration as it is not defined as in the official target configuration.

It really should be 

```
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1
```

haslinghuis, 2024-03-04:
Did you use save ? - as the support ID shows the difference.

fedezcason, 2024-03-04:
Yes i did the gyro status is still the same

haslinghuis, 2024-03-04:
Submit new support ID here so we can see the change

fedezcason, 2024-03-04:
cb6c8998-9219-41c4-880c-302c7f6076dd there it is.
Also i just noticed that i don't have the OSD section in the left buttons, do you know why?

haslinghuis, 2024-03-04:
Still see

```
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
```

Also resources and configuration does not make sense:

- https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-HGLRCF722MINI.config
- https://build.betaflight.com/api/support/32a1f394-fe55-4800-a782-2b2e13724caf

Did you flash recently and use full chip erase while doing so?

WRONG TARGET - Seems you need HGLRCF722MINIV2

fedezcason, 2024-03-04:
So what i have to do? i flashed with MINIV2 but error still there

haslinghuis, 2024-03-04:
Full chip erase enabled ?

fedezcason, 2024-03-04:
Nope

fedezcason, 2024-03-04:
I have to?

haslinghuis, 2024-03-04:
Yes - this would explain the issue.

fedezcason, 2024-03-04:
i flashed with the erase.
now what i have to do?

haslinghuis, 2024-03-04:
Check status

fedezcason, 2024-03-04:
Still the same.

Here's the ID 13ad2c6e-577e-4e10-948b-31ac2c97d772

fedezcason, 2024-03-04:
the fact that the gyro says locked is not a problem?

haslinghuis, 2024-03-04:
This is good. It was shared before - now it is locked and using dma.

fedezcason, 2024-03-04:
thank you so much mate