betaflight/betaflight #13425: Building for MATEKH743 fails on macOS

valeriyvan, 2024-03-06:
### Describe the bug

Building for MATEKH743 fails on macOS

```
betaflight % make configs
make: Circular configs <- configs dependency dropped.
Hydrating clone for configs: ./src/config
git clone https://github.com/betaflight/config ./src/config
Cloning into './src/config'...
remote: Enumerating objects: 5852, done.
remote: Counting objects: 100% (31/31), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 5852 (delta 15), reused 10 (delta 2), pack-reused 5821
Receiving objects: 100% (5852/5852), 1.15 MiB | 5.24 MiB/s, done.
Resolving deltas: 100% (3050/3050), done.
betaflight % make TARGET=MATEKH743      
Makefile:131: src/main/target/MATEKH743/target.mk: No such file or directory
Makefile:166: *** No TARGET_MCU specified. Is the target.mk valid for MATEKH743?.  Stop.
```

### To Reproduce

`git clone https://github.com/betaflight/betaflight.git`
`cd betaflight`
`make configs`
`make TARGET=MATEKH743`

### Expected behavior

Building for MATEKH743 target should succeed on macOS

### Support ID

```txt
n/a
```

### Flight controller

n/a

### Other components

n/a

### How are the different components wired up (including port information)

n/a

### Add any other context about the problem that you think might be relevant here

Building for STM32H743 succeeds:

```
betaflight % make TARGET=STM32H743
/Applications/Xcode.app/Contents/Developer/usr/bin/make -j ./obj/betaflight_4.5.0_STM32H743.hex
rm -f ./obj/main/STM32H743/.efhash_*
EF HASH -> ./obj/main/STM32H743/.efhash_818b631372f10ad09e29463180b1c7d7
%% startup_stm32h743xx.s 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (speed optimised) ./src/main/drivers/bus_quadspi.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/stm32/audio_stm32h7xx.c 
%% (optimised) ./src/main/drivers/stm32/adc_stm32h7xx.c 
%% (size optimised) ./src/main/drivers/stm32/bus_i2c_hal_init.c 
%% (optimised) ./src/main/drivers/stm32/bus_i2c_hal.c 
%% (optimised) ./src/main/drivers/stm32/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/stm32/bus_quadspi_hal.c 
[... 414 lines cut ...]
%% (optimised) ./src/main/drivers/rx/rx_a7105.c 
%% (optimised) ./src/main/drivers/rx/rx_cyrf6936.c 
%% (optimised) ./src/main/drivers/rx/rx_sx127x.c 
%% (optimised) ./src/main/drivers/rx/rx_sx1280.c 
%% (optimised) ./src/main/drivers/stm32/vcp_hal/usbd_desc.c 
%% (optimised) ./src/main/drivers/stm32/vcp_hal/usbd_conf_stm32h7xx.c 
%% (optimised) ./src/main/drivers/stm32/vcp_hal/usbd_cdc_hid.c 
%% (optimised) ./src/main/drivers/stm32/vcp_hal/usbd_cdc_interface.c 
%% (optimised) ./src/main/drivers/stm32/serial_usb_vcp.c 
%% (optimised) ./src/main/drivers/usb_io.c 
%% (size optimised) lib/main/google/olc/olc.c 
Linking STM32H743 
Memory region         Used Size  Region Size  %age Used
           FLASH:         664 B       128 KB      0.51%
    FLASH_CONFIG:          0 GB       128 KB      0.00%
          FLASH1:      608487 B      1792 KB     33.16%
        ITCM_RAM:       19064 B        64 KB     29.09%
        DTCM_RAM:       23664 B       128 KB     18.05%
             RAM:      112564 B       512 KB     21.47%
          D2_RAM:          0 GB       256 KB      0.00%
       MEMORY_B1:          0 GB         0 GB
   text	   data	    bss	    dec	    hex	filename
 599551	   9600	 126620	 735771	  b3a1b	./obj/main/betaflight_STM32H743.elf
Creating HEX ./obj/betaflight_4.5.0_STM32H743.hex
```

bigspeedfpv, 2024-03-06:
try `make CONFIG=MATEKH743` - works for me 
from what I can tell this is a change for Unified Targets that was never documented

haslinghuis, 2024-03-06:
Just `make MATEKH743` would do.

closed 2024-03-06 by haslinghuis

valeriyvan, 2024-03-07:
> Just `make MATEKH743` would do.

That means developer documentation should be fixed.

valeriyvan, 2024-03-07:
Neither `make CONFIG=MATEKH743` nor `make MATEKH743` is mentioned in official documentation on page  https://betaflight.com/docs/development/building/Building-in-Mac-OSX. But it twice mentions `make TARGET=MATEKH743`. What's correct, canonical way of building,  `make CONFIG=MATEKH743` or `make MATEKH743`? I would like to fix documentation.