betaflight/betaflight #13428: My Gyro is not showing up - LUXF7HD

Shldxn, 2024-03-07:
### Describe the bug

My gyro is not appearing no matter my best efforts

### To Reproduce

I flashed it to get the servos tab on the beta flight, but when I flashed it the gyro is no longer showing up

### Expected behavior

I don't really know. Im new to this, this is my first drone

### Support ID

```txt
f3b0caa3-e816-41f0-ad8a-dab311cc6927
```

### Flight controller

Lumenier Lux F7 HD lite Flight Controller

haslinghuis, 2024-03-07:
@Shldxn Please perform a core only build and resubmit the support ID.

[image]

@nerdCopter do I see a bug here (are we supposed to have custom defines and select Pull Request or Commit fields here)?

To reproduce. Auto-detect -> select core only (fields are not visible) -> now pick another board -> fields show up :)

nerdCopter, 2024-03-07:
agree, bug. reproducible.

sugaarK, 2024-03-08:
reproducible for me to

nerdCopter, 2024-04-19:
@Shldxn never responded with core-build support ID.  or if the Guro model is known directly.  Only MPU6000 is defined in the target files.

nerdCopter, 2024-04-19:
```
from: GetFPV Support <support@getfpv.com>
1:10 PM (2 minutes ago)
to: nerdcopteranon
Apr 19, 14:09

Hello nerdcopter,

[...redacted off-subject text...]

The Lux F7 HD lite only uses the MPU6000. These are all Matek boards and there was no changes during the production run 
so there absolutely would not be any Lux F7s out there with a replacement gyro. Any found would not be legitimate. No BMI 
or ICM swaps in our production run.

Best Regards,
Brian

--
Brian Sliker
Customer Service Representative / Return Specialist
```

closed 2024-04-19 by nerdCopter