betaflight/betaflight #13433: 4.5RC2 OSD stuck after "Save & reboot" joystick command

HishamGhosheh, 2024-03-10:
### Describe the bug

Started to have this issue after upgrading to 4.5RC2
Even if I don't do any modifications it still gets stuck on "Rebooting...". The quad can arm and fly and everything but OSD doesn't update unless I power cycle. USB is not connected.

https://github.com/betaflight/betaflight/assets/5965360/01f4fb0d-e062-4b97-98ee-908cad91f51f

[image]

Another thing that might be related: No OSD until the quad stops beeping and the RC is connected

### To Reproduce

Open the OSD menu with RC
Go to Save & Reboot
Reboot the quad

### Expected behavior

OSD goes out of the menu and returns to whatever profile was selected before.

### Support ID

```txt
8f63126d-d1a3-41dc-8b8e-c731306092a1
```

### Flight controller

SpeedyBee F405 V3 Stack BLS 50A 30x30 FC&ESC iNAV Betaflight Blackbox

### Other components

AKK FX2 Ultimate Mini 5.8GHz 40CH 25mW/200mW/600mW/1200mW 
RunCam Phoenix 2 SP Camera DC 5-36V Screen Ratio 4:3/16:9 7.5g 19*19*21mm 

### How are the different components wired up (including port information)

UART1 -> TBS Smartaudio

HishamGhosheh, 2024-03-16:
Searching online for this reveals it's an issue with SpeedyBee F405 when an SD card is present

haslinghuis, 2024-03-16:
This board shares SPI [between OSD and blackbox] as this [could] lead to dead locks in firmware.

closed 2024-03-16 by haslinghuis