betaflight/betaflight #13443: Safety GPS home point

ASDosjani, 2024-03-12:
### Is your feature request related to a problem? Please describe

Main idea behind this feature are unstable gps locks. When I want to start the flight from a closed space and the gps home point is unstable inside. Sometimes I arm without enough satellites because the second before arm one satellite drops.

### Describe the solution you'd like

My idea is to save the coordinates when the GPS locks to 4 or 5 satellites (I don't know the exact minimum) then update this safety home point until takeoff. If the signal is lost the drone still can fly home instead of dropping.

OSD Warning message could be changed something like "Inaccurate home point" when taking off with less than the set up minimum satellite count.

### Describe alternatives you've considered

-

wterreb, 2024-04-06:
Love this idea.   Also if you always fly at the same spot and those take-off coordinates is pre-saved, then even if you started flying without a GPS fix, then at least you have a chance of a proper return to home if the GPS managed to get a fix mid flight.    Maybe it can just always save the last coordinates of an arming with proper GPS fix and then use that if you took off without a fix.   

Of course there is also the danger of a fly-away if you happened to fly at a different spot that time!  But it could be something you could opt in with a tickbox.

ASDosjani, 2024-04-06:
Saving the last coordinates wouldn't be safe, but using manual home point is a good idea. Maybe a pre arm confirmation would be nice for the manual home point.

ctzsnooze, 2024-04-06:
I don't think we currently intend any manual configuration of the home point.
We do not want to encourage people taking off in the 'hope' that later they get some random GPS position (which may be very inaccurate, or over a populous area, or a road) that they will then fly back to in failsafe.  There are serious safety issues.

closed 2024-04-06 by ctzsnooze

ASDosjani, 2024-04-06:
@ctzsnooze what about my original idea? When the lock is unstable because the gps signal is blocked by buildings or trees smth. But after you fly in the air it becomes stable and could initiate gps rescue even if you armed without lock because it was dropped.
Most of the times people just need a quad which comes back in the right direction so the video signal can come back and they can land manually.

ctzsnooze, 2024-05-14:
Why not set the min required sats to 5?

ASDosjani, 2024-05-14:
> Why not set the min required sats to 5?

Already on 5