betaflight/betaflight #13446: BMP390 Barometer Support

WyattSmithTritium, 2024-03-12:
### Is your feature request related to a problem? Please describe

The BMP388 is obsolete ([link1](https://www.bosch-sensortec.com/products/environmental-sensors/pressure-sensors/bmp388/), [link2](https://www.digikey.com/en/products/detail/bosch-sensortec/BMP388/8322638)) and the recommended upgrade is the BMP390.

### Describe the solution you'd like

Let's add support for the BMP390 using the BMP388 implementation as a starting point.

### Describe alternatives you've considered

We've considered switching to the BMP280, but it is also obsolete.

### Other information

This [repo](https://github.com/boschsensortec/BMP3_SensorAPI/tree/master) can be used as reference.

WyattSmithTritium, 2024-03-12:
I have hardware available to help test this feature. I can also help with the implementation if devs are too busy.

ledvinap, 2024-03-12:
IMO all that is necessary is to add new device id (BMP388_DEFAULT_CHIP_ID). And maybe update measurement times.

ws998116, 2024-03-12:
I agree @ledvinap! Can we add BMP390 support to the existing BMP388 files, or should the files be duplicated and edited for the BMP390?

WyattSmithTritium, 2024-03-12:
I agree @ledvinap! Can we add BMP390 support to the BMP388 files, or should they be duplicated for the BMP390?

haslinghuis, 2024-03-12:
Look here for an example for adding a new device ID: https://github.com/betaflight/betaflight/pull/13427

ledvinap, 2024-03-12:
Actually, only https://github.com/betaflight/betaflight/pull/13427/files#diff-6ba516408cf301acec6289be3f710bc081f212cc44c18f6f5496dd1473cc7de0L305-R323 is relevant.

ws998116, 2024-03-18:
Resolved with #13458. Good to close this issue

closed 2024-03-19 by haslinghuis