betaflight/betaflight #13456: Position hold and optical flow

Delvien, 2024-03-15:
### Is your feature request related to a problem? Please describe

The problem is having to use iNAV for these features. :(

### Describe the solution you'd like

Implement position hold and optical flow (lidar sensors like the MFT-01) 

### Describe alternatives you've considered

Using iNAV, and its painful.  I'd rather use betaflight.

### Other information

I think this would benefit all users looking to use Betaflight for their long range quads, and I heard BF is doing fixed wings soon? So this could assist with the feature-set there too. 

Y'all do an awesome job, keep up the great work. Thank you for your consideration.

ctzsnooze, 2024-04-06:
Yes, it would be nice.

closed 2024-04-06 by ctzsnooze

Delvien, 2024-04-06:
Why was this closed? Is it approved or denied or... ?

haslinghuis, 2024-04-06:
Position hold is in the cards. Optical flow could be next.

Delvien, 2024-04-06:
Awesome, thank you.

CorvusMonitoring, 2024-05-28:
@haslinghuis , would it be already a piece of code to test OF and rangefinder?
We have 6 custom whoop equipped with that run under arducopter, but there is a lot of limitation in flight modes and tuning...
We would be honored to test it under betaflight