betaflight/betaflight #13461: M10 GPS Dilution of precision (DOP) worse on 4.5 than 4.4

Giammi, 2024-03-19:
### Is your feature request related to a problem? Please describe

I have several quads, all of them equipped with M10 GPS (Foxeer M10 250). Both in 4.4 and 4.5 they can reach up to over 20 satellites, but DOP is much much worse in 4.5 compared to 4.4 (with exact same configuration and within minutes from one flight to the other).

For instance:
- BF 4.4: I can easily get 8-10 sats in few seconds from a cold start and have a DOP of  around 1.0 (enough to arm safely). In flight the number of sats goes up to 25 with a DOP of about 0.5-0.6;
- BF 4.5 (all RC): I still get 8-10 sats in few seconds, but DOP is now 3-5 (not safe to arm with this home point). In flight the number of sats reaches up to 20-23 max (not 25, but ok), but DOP is now "only" 1.2-1.4, worse than BF 4.4 with just 10 sats.

In both cases GPS configured as follow (Italy based):
- Baud rate 57.600 as specified in the GPS Rescue guide
- set gps_sbas_mode = EGNOS
- set gps_ublox_use_galileo = ON
- set gps_set_home_point_once = ON

Hardware used:
- Radix2 + Foxeer M10 250
- Kakute H7 + Foxeer M10 250
- Argus Pro + Foxeer M10 250

I post a couple of screenshots from my DVRs just to show what I mean. They're not taken from two consequent flights though.

BF 4.4
[image]

BF 4.5
[image]

### Describe the solution you'd like

DOP to be consistent between 4.4 and 4.5, being about 0.5 with >20 sats.

### Describe alternatives you've considered

No alternatives found

Giammi, 2024-03-23:
I add a screenshot from today. 10 satellites locked and HDOP = 3.2. As a result, the home arrow was way off even after flying straight for 100.

[image]

Giammi, 2024-03-23:
Created as ISSUE #13476

ctzsnooze, 2024-03-23:
Hi.  In 4.5, if auto-configuration is enabled, we request data from any M8 or higher module in the NAV_PVT message.  That message provides pDOP values, and that value is what you see in the OSD.  I have checked the code very carefully, and that is how it is.

We do not receive hDOP or vDOP in 4.5, unless the GPS Module is not responding to auto-configuration, or is not being auto-configured, and is sending either NMEA packets or other packets (there is a specific DOP packet that includes all three DOP values).  

There is a debug mode which you can set, `GPS_DOP`, which will show in Sensors tab, or in a log, each of the three possible values, and satellite count.  While connected to Configurator, you can see these values in the Sensors tab.  They will tell you what values you are receiving.  In 4.5, with normal auto configuration, you should only see a number for pDOP.  

```
    DEBUG_SET(DEBUG_GPS_DOP, 0, gpsSol.numSat);
    DEBUG_SET(DEBUG_GPS_DOP, 1, gpsSol.dop.pdop);
    DEBUG_SET(DEBUG_GPS_DOP, 2, gpsSol.dop.hdop);
    DEBUG_SET(DEBUG_GPS_DOP, 3, gpsSol.dop.vdop);
```

The same debug exists in 4.4, so you could compare 4.4 to 4.5.

The code in 4.4 may, or may not, depending on the GPS configuration, have all three DOP estimates.  If you see a DOP value in the OSD, it will be `hDOP`.

`hDOP` only considers horizontal accuracy, and that is typically better than vertical accuracy.  Hence the values for `vDOP` (vertical accuracy) or `pDOP` (3D accuracy) are typically not as good as `hDOP`.

It is likely that your 4.4 module is returning all three values, and that you are seeing `hDOP`, which will give a 'better' estimate of precision than `pDOP` in 4.5.

The reasons  that we do not request all three DOP values in 4.5 are:
- to get all three DOP values requires an additional DOP data packet, and that overhead is considerable
- `pDOP` is very closely related to `hDOP`, and is a useful estimate of overall precision
- we have aligned the estimate of GPS trust to DOP so that it works right with `pDOP`
- in 4.5, auto-config will ensure that we always get `pDOP`, whereas in 4.4 it was a bit of random luck if you got hDOP, depending on module configuration.

I would recommend that you get accustomed to the new DOP value.  It is numerically different from what you previously saw.  

I should also note that M10 will quickly report a lot of sats after powering up, but often poor DOP until some time later on.  If you watch the reported speed, altitude, latitude and longitude values during sat acquisition, as more sats become available, those values stabilise better, and the degree of stability is well reflected in the DOP value shown in the OSD.

This is something I will explain better in the documentation.

Giammi, 2024-03-25:
Thanks @ctzsnooze! You couldn't have been more clear.

At this point, based on your experience, what do you consider a good pDOP to safely takeoff? In the past, with only HDOP as a reference, I normally used < 2 as a rule of thumb. And I could achieve that with 8-9 sats.

I ask this, because, referring to my comment above about home arrow pointing wrong direction, now I don't now if this has to be addressed to a bad DOP (if it is bad at all, since maybe HDOP is very good) or to something else. I guess at this point is something else, becasue the compass is WAY off. For instance, FYI, during this flight I had the magnetometer disabled from configurator, even if it was fully configured to work properly (I disabled it because I am also experiencing the issue #13360). I say this because I also left GPS_RESCUE_USE_MAG = ON and I am wondering if in such case should I change it to OFF or the system automatically relies on GPS or MAG based on the configuration.

ctzsnooze, 2024-04-06:
Home arrow is not affected by DOP.

Home position is calculated from home position vs current position (coordinates) of the quad.  At say 100m distance, even 3-5m position error causes very little error in the estimate of the angle to home.

Current heading estimate of the nose of the quad in GPS-only flight is based on the reported GPS heading (vector that the module is travelling in, over ground), *on the assumption that the nose of the quad is pointing in the same direction*. This assumption is false when the quad moves slowly and/or is drifted by wind, or drifts/flies backwards or sideways.

The arrow is calculated as difference between GPS course over ground vector and angle to home.

Arrow angle is influenced by 'drift' sideways since it is assumed that the quad is moving nose first.  

At distances of 100ms or so, and spreads of >5m/s, flying nose forward, on days without wind, arrow should be good and stable with current nightly builds using GPS only.

Please only use the latest nightly firmware where we have improved heading estimate accuracy.

Home arrow is BADLY affected by using older firmware and by having Mag enabled but not working accurately.

closed 2024-04-06 by ctzsnooze