betaflight/betaflight #13466: GyroUnfilt and Gyro_Scaled data mismatch

jesuslg123, 2024-03-21:
### Describe the bug

Exploring a blackbox CSV file I have noticed that the values for the debug value `GYRO_SCALED` and the new fields `gyroUnfilt` are different.

As far as I know `gyroUnfilt` is the new place for `GYRO_SCALED` debug value, so it should contain the same value.

[image]

[4.5.bbl.remove.txt](https://github.com/betaflight/betaflight/files/14693897/4.5.bbl.remove.txt)
* Is bbl extension.

### To Reproduce

- Setup a FC with latest 4.5 firmware version
- Enable blackbox debug mode with `gyro_scaled`
- Make a flight test
- Extract the blackbox file
- Compare data
- `gyroUnfilt` and `debug` is not the same data

### Expected behavior

`gyroUnfilt` fields should have the same values as `debug` fields

### Support ID

```txt
No available.
```

### Flight controller

Custom Flight Controller with F435

### Other components

Gyro is MPU6000

haslinghuis, 2024-03-21:
Does it have to do with downsampling vs averaging? Try disabling gyro lowpass 2

```bash
bool downsampleFilterEnabled;      // if true then downsample using gyro lowpass 2, otherwise use averaging
```

```bash
{"gyroUnfilt",  0, SIGNED,   .Ipredict = PREDICT(0),       .Iencode = ENCODING(SIGNED_VB),   .Ppredict = PREDICT(AVERAGE_2),     .Pencode = ENCODING(SIGNED_VB), CONDITION(GYROUNFILT)},
{"gyroUnfilt",  1, SIGNED,   .Ipredict = PREDICT(0),       .Iencode = ENCODING(SIGNED_VB),   .Ppredict = PREDICT(AVERAGE_2),     .Pencode = ENCODING(SIGNED_VB), CONDITION(GYROUNFILT)},
{"gyroUnfilt",  2, SIGNED,   .Ipredict = PREDICT(0),       .Iencode = ENCODING(SIGNED_VB),   .Ppredict = PREDICT(AVERAGE_2),     .Pencode = ENCODING(SIGNED_VB), CONDITION(GYROUNFILT)},
```

```bash
// DEBUG_GYRO_SCALED records the unfiltered, scaled gyro output
// If downsampling than the last value in the sample group will be output
GYRO_FILTER_DEBUG_SET(DEBUG_GYRO_SCALED, axis, lrintf(gyro.gyroADC[axis]));
```
> maybe just data is written in different moment so data was modified

tbolin, 2024-03-25:
Copying from discord so it's visible here too
> gyro.gyroADC gets updated at gyro native rate in gyroUpdate, called from taskGyroSample
the value in the GYRO_SCALED debug field is updated in filterGyroDebug, called from gyroFiltering which is again called from taskFiltering
Both the gyro.gyroADC and debug values are pulled in to the blackbox main state in subTaskPidSubprocess, called at the end of taskMainPidLoop.
If taskGyroSample runs between taskFiltering and taskMainPidLoop, that might explain the difference

closed 2024-05-02 by github-actions[bot]