betaflight/betaflight #13482: Add support for the STM32H5 series MCUs

hydra, 2024-03-29:
### Is your feature request related to a problem? Please describe

The H5 MCUs, specifically the STM32H562RG, makes a good candidate for a high performance, and cheap, flight controller MCU with up-to-date peripherals.

Work is already underway, this is just to track progress.

[ ] Updated HAL libs (https://github.com/betaflight/betaflight/pull/13442)
[ ] Build system support.
[ ] Linker script
[ ] CPU startup code (assembler + basic clock configuration)
[ ] Minimal H5 target (with lots of #undefs) (no drivers, no scheduler tasks, just something that boots)
[ ] UART support
[ ] USB support
[ ] SPI support
[ ] PWM support
[ ] ADC support
[ ] DShot support

Please link PR's as they're raised.

Anyone interested in FC hardware prototypes please contact me via the SPRacing or BF discord servers.

H5 series overview: https://www.st.com/en/microcontrollers-microprocessors/stm32h5-series.html

Note there are two types of H5 MCU:
* H50x - low-cost ones, small flash, low ram, good for motor control, etc.
* H56x - better, large flash, plenty of ram, perfect for FCs.  The STM32H562RGT6 is perfect for small FCs

The H562 is essentially the same as the H563 but without some peripherals that BF doesn't use anyway, so the H563 makes for a good development system as code will run on the H562 as well.

H563 Nucleo: https://www.st.com/en/evaluation-tools/nucleo-h563zi.html

Datasheet: [stm32h562rg.pdf](https://github.com/betaflight/betaflight/files/14804033/stm32h562rg.pdf)
Reference manual: https://www.st.com/resource/en/reference_manual/rm0481-stm32h563h573-and-stm32h562-armbased-32bit-mcus-stmicroelectronics.pdf

block diagram:

[image]

system architecture:

[image]

mouser links:

https://www.mouser.sg/ProductDetail/STMicroelectronics/STM32H562RGT6?qs=amGC7iS6iy80F6W31pzhsQ%3D%3D (in stock and shipping, as of 2024/03/29)

Of note to developers, the H56x has an ETM so it can be used to do real-time system profiling using a cheap debugger.  Tools like segger SystemView and O-Zone can be used to see which functions are using the CPU the most.
Other highlights include OctoSPI, USBC-PD, I3C and an updated DMA controller.

### Describe the solution you'd like

Flying BF with an H5 😄 

### Describe alternatives you've considered

Other MCU from other manufacturers

alex39292, 2024-03-30:
This is a good idea to use H5 series microcontroller! I'm with you!💪

iottrends, 2024-07-02:
Any updates on this!! @hydra, I would like to contribute... it can be porting or designing  FC or both...

BR
Abhinav

ggabriel-airmod, 2025-01-24:
same here, @hydra, i have to contribute on this.
My company wants to develop a new flight controller using the STM32H562.
So, without a firmware, we will have a shiny but useless piece of plastic.

hydra, 2025-04-07:
@iottrends @ggabriel-airmod I'm not working on this myself, but others have made progress since there are related commits. I'm not sure of the status of H5 support, if people working on it could make themselves known here and update the TODO list above that'd be great.

closed 2026-09-26 by blckmn