betaflight/betaflight #13483: 4.5.0 RC3 compile error if SPEKTRUM radio protokol is enabled

AnDann, 2024-03-29:
### Describe the bug

If rx protocol SPEKTRUM is selected in Firmware Flasher for 4.5.0 RC3 I get the following compile error

`In file included from ./src/main/pg/rx.c:29:
./src/main/pg/rx.c: In function 'pgResetFn_rxConfig':
./src/main/pg/rx.c:48:27: error: 'SERIALRX_SPEKTRUM' undeclared (first use in this function); did you mean 'SERIALRX_SPEKTRUM1024'?
   48 | #define SERIALRX_PROVIDER SERIALRX_SPEKTRUM
      |                           ^~~~~~~~~~~~~~~~~
./src/main/config/config_reset.h:41:9: note: in definition of macro 'RESET_CONFIG_2'
   41 |         __VA_ARGS__                                       \
      |         ^~~~~~~~~~~
./src/main/pg/rx.c:80:30: note: in expansion of macro 'SERIALRX_PROVIDER'
   80 |         .serialrx_provider = SERIALRX_PROVIDER,
      |                              ^~~~~~~~~~~~~~~~~
./src/main/pg/rx.c:48:27: note: each undeclared identifier is reported only once for each function it appears in
   48 | #define SERIALRX_PROVIDER SERIALRX_SPEKTRUM
      |                           ^~~~~~~~~~~~~~~~~
./src/main/config/config_reset.h:41:9: note: in definition of macro 'RESET_CONFIG_2'
   41 |         __VA_ARGS__                                       \
      |         ^~~~~~~~~~~
./src/main/pg/rx.c:80:30: note: in expansion of macro 'SERIALRX_PROVIDER'
   80 |         .serialrx_provider = SERIALRX_PROVIDER,
      |                              ^~~~~~~~~~~~~~~~~
make[2]: *** [Makefile:439: obj/main/STM32F7X2_SKYSTARSF7HDPRO/pg/rx.o] Error 1
make[2]: *** Waiting for unfinished jobs....
%% (optimised) ./src/main/pg/stats.c 
%% (optimised) ./src/main/pg/timerio.c 
make[2]: Leaving directory '/source'
make[1]: *** [Makefile:559: hex] Error 2
make[1]: Leaving directory '/source'
make: *** [mk/config.mk:62: SKYSTARSF7HDPRO] Error 2
make failed with 2`

I think it is related to commit #13371

### To Reproduce

- Go to firmware flasher select release 4.5.0 RC3
- Activate RX protocol SPEKTRUM

### Expected behavior

Should compile like 4.5.0 RC2 does

### Support ID

```txt
ID: "939a418c-379d-4203-8705-0c16fb6a316d"
```

### Flight controller

SKYSTARSF7HDPRO

haslinghuis, 2024-03-29:
- Confirm it does not work with 4.5.0-RC3
- Tested local build on master without problem with following command:

```
make SKYSTARSF7HDPRO EXTRA_FLAGS="-DCLOUD_BUILD -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_MAG -DUSE_OSD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_SPEKTRUM -DUSE_VTX"
```

Cloud build using configurator:

- [FAIL] 4.5 RC3 log: https://build.betaflight.com/api/builds/e26afb4e23372e0513af1eb5fc0027cc/log
- [OK] 4.5.0-dev log: https://build.betaflight.com/api/builds/1812542d955b5d83ff25e4b95c6f82ba/log

ror1948, 2024-04-07:
ya same bug for me as well

haslinghuis, 2024-04-07:
- Fixed in #13459

closed 2024-04-07 by haslinghuis

ror1948, 2024-04-07:
still fails for me ..tried several times

make[2]: *** [Makefile:440: obj/main/STM32F405_SPEEDYBEEF405MINI/pg/rx.o] Error 1
make[2]: *** Waiting for unfinished jobs....
%% (optimised) ./src/main/pg/timerup.c 
make[2]: Leaving directory '/source'
make[1]: *** [Makefile:559: hex] Error 2
make[1]: Leaving directory '/source'
make: *** [mk/config.mk:62: SPEEDYBEEF405MINI] Error 2
make failed with 2
 ***** Completed Build *****
Build time: 2024-04-07 21:07:13Z, time taken: 13.39
Submitted:  2024-04-07 21:06:59Z, wait time: 14.57

board speedybeeF405mini

ror1948, 2024-04-07:
[buildfail.txt](https://github.com/betaflight/betaflight/files/14899005/buildfail.txt)
my build fail log file

haslinghuis, 2024-04-07:
<details>
  <summary><i>Works here</i></summary>
  
```
mark@NUC12WSHi7:~/dev/betaflight/betaflight/clang-test/betaflight$ make SPEEDYBEEF405MINI EXTRA_FLAGS="-D'BUILD_KEY=cc5852bc23b5a3bdcafc6cbcbbc67c01' -D'RELEASE_NAME=4.5.0-RC3' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_SPEKTRUM -DUSE_VTX"
Building target config SPEEDYBEEF405MINI
make -j hex CONFIG=SPEEDYBEEF405MINI
make[1]: Entering directory '/home/mark/dev/betaflight/betaflight/clang-test/betaflight'
make -j ./obj/betaflight_4.5.0_STM32F405_SPEEDYBEEF405MINI.hex
make[2]: Entering directory '/home/mark/dev/betaflight/betaflight/clang-test/betaflight'
rm -f ./obj/main/STM32F405_SPEEDYBEEF405MINI/.efhash_*
EF HASH -> ./obj/main/STM32F405_SPEEDYBEEF405MINI/.efhash_c7208ac9e07ffc6188cf6c1887a9f402
%% startup_stm32f40xx.s 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/stm32/pwm_output_dshot.c 
%% (optimised) ./src/main/drivers/stm32/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/stm32/bus_i2c_stm32f4xx.c 
%% (optimised) ./src/main/drivers/stm32/bus_spi_stdperiph.c 
%% (optimised) ./src/main/drivers/stm32/debug.c 
%% (optimised) ./src/main/drivers/stm32/dma_reqmap_mcu.c 
%% (optimised) ./src/main/drivers/stm32/dma_stm32f4xx.c 
%% (optimised) ./src/main/drivers/stm32/dshot_bitbang_stdperiph.c 
%% (optimised) ./src/main/drivers/stm32/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/stm32/exti.c 
%% (optimised) ./src/main/drivers/stm32/io_stm32.c 
%% (optimised) ./src/main/drivers/stm32/light_ws2811strip_stdperiph.c 
%% (optimised) ./src/main/drivers/stm32/persistent.c 
%% (optimised) ./src/main/drivers/stm32/pwm_output.c 
%% (optimised) ./src/main/drivers/stm32/rcc_stm32.c 
%% (optimised) ./src/main/drivers/stm32/sdio_f4xx.c 
%% (optimised) ./src/main/drivers/stm32/serial_uart_stdperiph.c 
%% (optimised) ./src/main/drivers/stm32/serial_uart_stm32f4xx.c 
%% (optimised) ./src/main/drivers/stm32/system_stm32f4xx.c 
%% (optimised) ./src/main/drivers/stm32/timer_stdperiph.c 
%% (optimised) ./src/main/drivers/stm32/timer_stm32f4xx.c 
%% (optimised) ./src/main/drivers/stm32/transponder_ir_io_stdperiph.c 
%% (optimised) ./src/main/drivers/stm32/usbd_msc_desc.c 
%% (optimised) ./src/main/drivers/stm32/camera_control.c 
%% (optimised) ./src/main/drivers/flash.c 
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% arm_bitreversal2.S 
%% (optimised) ./src/main/drivers/flash_m25p16.c 
%% (optimised) ./src/main/drivers/flash_w25n01g.c 
%% (optimised) ./src/main/drivers/flash_w25q128fv.c 
%% (optimised) ./src/main/drivers/flash_w25m.c 
%% (optimised) ./src/main/io/flashfs.c 
%% (optimised) ./src/main/drivers/stm32/usb_msc_f4xx.c 
%% (optimised) ./src/main/drivers/usb_msc_common.c 
%% (optimised) ./src/main/msc/usbd_storage.c 
%% (optimised) ./src/main/msc/usbd_storage_emfat.c 
%% (optimised) ./src/main/msc/emfat.c 
%% (optimised) ./src/main/msc/emfat_file.c 
%% (optimised) ./src/main/msc/usbd_storage_sd_spi.c 
%% (optimised) ./src/main/msc/usbd_storage_sdio.c 
%% (optimised) ./src/main/drivers/sdcard.c 
%% (optimised) ./src/main/drivers/sdcard_spi.c 
%% (optimised) ./src/main/drivers/sdcard_sdio_baremetal.c 
%% (optimised) ./src/main/drivers/sdcard_standard.c 
%% (optimised) ./src/main/io/asyncfatfs/asyncfatfs.c 
%% (optimised) ./src/main/io/asyncfatfs/fat_standard.c 
%% (optimised) ./src/main/build/build_config.c 
%% (optimised) ./src/main/build/debug.c 
%% (optimised) ./src/main/build/debug_pin.c 
%% (optimised) ./src/main/build/version.c 
%% (size optimised) ./src/main/main.c 
%% (optimised) ./src/main/pg/adc.c 
%% (optimised) ./src/main/pg/beeper.c 
[... 353 lines cut ...]
%% (optimised) ./src/main/drivers/stm32/vcpf4/usbd_usr.c 
%% (optimised) ./src/main/drivers/stm32/vcpf4/usbd_cdc_vcp.c 
%% (optimised) ./src/main/drivers/stm32/vcpf4/usb_cdc_hid.c 
%% (optimised) ./src/main/drivers/stm32/serial_usb_vcp.c 
%% (optimised) ./src/main/drivers/usb_io.c 
%% (size optimised) lib/main/google/olc/olc.c 
Linking STM32F405_SPEEDYBEEF405MINI 
Memory region         Used Size  Region Size  %age Used
           FLASH:         392 B        16 KB      2.39%
    FLASH_CONFIG:          0 GB        16 KB      0.00%
          FLASH1:      402765 B       992 KB     39.65%
   SYSTEM_MEMORY:          0 GB        29 KB      0.00%
             RAM:       75916 B       128 KB     57.92%
             CCM:       15108 B        64 KB     23.05%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
       MEMORY_B1:          0 GB         0 GB
   text	   data	    bss	    dec	    hex	filename
 397189	   5968	  84660	 487817	  77189	./obj/main/betaflight_STM32F405_SPEEDYBEEF405MINI.elf
Creating HEX ./obj/betaflight_4.5.0_STM32F405_SPEEDYBEEF405MINI.hex 
make[2]: Leaving directory '/home/mark/dev/betaflight/betaflight/clang-test/betaflight'
make[1]: Leaving directory '/home/mark/dev/betaflight/betaflight/clang-test/betaflight'
Building target config SPEEDYBEEF405MINI succeeded.
```

</details>

ror1948, 2024-04-07:
I talked to another guy in our club and it did not work for him either.Do not know why nor do I have the knowledge to troubleshoot.

thanks

ror1948, 2024-04-07:
What do I have to do to ensure a clean build with all new files.?

thanks

haslinghuis, 2024-04-07:
[image]

ror1948, 2024-04-07:
ok that workedthanks

 
 

image.png (view on web)