betaflight/betaflight #13484: MSP GPS does not show number of GPS Satellites in OSD

marcaregan, 2024-03-29:
### Describe the bug

The OSD does not show the correct number of GPS satellites that are locked when using a Flite Test EZ ID module with the "MSP" GPS protocol. 

### To Reproduce

When using the Flite Test EZ ID module, with it's TX connected to RX on the flight controller (FC), and after setting the GPS protocol to "MSP",  the FC recognizes the GPS module as well as the number of satellites in the "GPS" tab. However, the detailed "GPS Signal Strength" information is not shown in the "GPS" tab (i.e., locked satellites, signal strength, etc.). Moving on to the "OSD" tab, when selecting to show "GPS sats", the correct value does not show up in the OSD. Instead, the letters "NC" are shown, regardless of the number of satellites recognized by Betaflight under the "GPS" tab. 

[image]
[image]
[image]

### Expected behavior

OSD "GPS sats" should show the same number of satellites as the "GPS" tab does.

### Support ID

```txt
Id: "6780817b-8e4a-41bd-a51b-75612ed94447"
```

### Flight controller

SpeedyBee F7 V3 BL32 50A 30x30 Stack

### Other components

RadioMaster RP3 2.4Ghz
Flite Test EZ ID module
DJI O3 Air Unit

### How are the different components wired up (including port information)

RadioMaster RP3 2.4Ghz --> UART2
Flite Test EZ ID module --> UART6 (RX only)
DJI O3 Air Unit --> UART1 (clipped SBUS wire)

### Add any other context about the problem that you think might be relevant here

I'm wondering if the fact that the detailed "GPS Signal Strength" is not working for the MSP protocol GPS has something to do with the OSD showing "NC" for number of locked satellites. 

Here's an example of a different quad of mine that has a GPS that uses the UBLOX protocol, giving some early "GPS Signal Strength" data
[image]

ctzsnooze, 2024-04-06:
The Ublox protocol is fairly well defined and standardised.
There is no such standard protocol for MSP-connected GPS units.
We provide support for satellite information - `SVINFO` - over MSP, but it's up to the manufacturers to supply that data from their devices, and to ensure it works:
[image]
I suggest contacting the manufacturer, telling them it doesn't work, and asking them to fix it.

marcaregan, 2024-04-10:
Thank you for the information. I would be happy to reach out to the manufacturer to see if they can get the right information sent across MSP. However, I think I need to better understand the issue before I can properly formulate the suggestion to the manufacturer. 

I think where I am caught up is that I can see in the Betaflight configurator how many satellites the MSP-connected GPS has. Doesn't that mean the that number of satellites is getting passed from the MSP-connected GPS units, to the flight controller, and then to the Betaflight configurator? If so, shouldn't the OSD (which is being drawn by the Flight Controller) be able to show the correct number of GPS satellites that are locked?

closed 2024-05-19 by github-actions[bot]

Jamostaged225, 2025-01-02:
The reason this is likely happening is because your current flight controller has a build in barometer. Disable both the “Barometer” and “Magnetometer” under the configuration tap. This should now utilize the other GPS being passthrough.

TWrecX916, 2025-03-02:
Please open this back up. This is a bug!!!!

When using a REMOTE ID module for GPS (I'm using the phoenix MRID), you have to be in MSP mode, for the FC to talk to the remote ID module, it then talks to the GPS in ublox. 

ALL OF THIS WORKS WELL! BUT!!!! 

If you have a barometer turned on, 
it still works well except the OSD satellites show NC.