betaflight/betaflight #13485: Mavlink Telemetry does not work

catkira, 2024-03-29:
### Describe the bug

When I configure a port for mavlink telemetry output nothing comes out. I also tried to enable the telemetry mode, that did not work either.

### To Reproduce

put an oscilloscope on the configured uart

### Expected behavior

see some bits on oscilloscope

### Support ID

```txt
8cdc29a0-6bd0-4500-835c-6343936bf5ba
```

### Flight controller

BetaFPV Toothpick F722 AIO Flight Controller 35A

catkira, 2024-03-30:
For me the flasher screen looks like the image below. Does that mean I cannot enable mavlink telemetry if I use CRSF for rc control?
[image]

catkira, 2024-03-30:
I see the cloud build that I used was using this command line for make
```
 make BETAFPVF722 EXTRA_FLAGS="-D'BUILD_KEY=14cac3e8314c9e55edd1650bdc69ae22' -D'RELEASE_NAME=4.5.0-RC3' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"
```
It does not change even when I add TELEMETRY_MAVLINK in the flasher GUI as custom define. Does that mean I have to compile locally with this custom define? And is it possible to have CRSF and MAVLINK telemetry enabled at the same time?
I want to have both,  CRSF telemetry via ELRS and MAVLINK telemetry via UART.

haslinghuis, 2024-03-30:
Just tested a 4.5.0-dev build and added `TELEMETRY_MAVLINK` using custom defines in configurator 10.10.0 [master] and it shows up in the build log:

https://build.betaflight.com/api/builds/cd5b82542e1c6e778ac1f7abec186678/log

catkira, 2024-04-03:
ah ok thanks for the info. I did a custom cloud build with your define, but it still doesn't work. Do I have to do anything more besides enabling the telemetry mode to get telemetry output?

haslinghuis, 2024-04-03:
Yes enabling Mavlink telemetry in ports tab should be enough. There is not much documentation about it's use in Betaflight:

https://betaflight.com/docs/development/Telemetry#mavlink-telemetry

Also unaware about the usage of Mavlink with CRSF telemetry already running. Did you try disabling telemetry on receiver tab.
I would have need to look at the code to find out more clear answers.

catkira, 2024-04-03:
I tried to disable telemetry for the receiver, it's still not working. This is not a high priority problem for me now, because I just enabled another MSP port and read out my telemetry from there. It's only slightly more inefficient, because I have to do polling. Or is it possible to subscribe to telemetry values over the MSP protocol?

haslinghuis, 2024-04-03:
Not using topics to subscribe here - so polling is the way to go as alternative. Still as mavlink is hardly used with BF - Mavlink code has not been maintained over time - I'm unable to verify functionality without specific hardware config.

SerJG, 2024-07-26:
@catkira Hello, I have the same issue - no Mavlink data out of UART from my FC. 
Have you found a way to enable the Mavlink telemetry feature properly?

CurryKitten, 2024-08-27:
I've also found this issue, I did try to flash with a custom define of TELEMETRY_MAVLINK which didn't work and also resulted in the very strange situation of seeing 2 telemetry options in the receiver tab

[image]

@catkira Perhaps you could reopen, or document your workaround?  In my case I'm using an OpenIPC camera/groundstation which uses Mavlink to populate it's OSD.  I only mention it as it's quite probably not as flexible as Ardupilot might be in any workaround

catkira, 2024-08-27:
I dont have a workaround, except reading data out via MSP

seriyps, 2024-09-19:
I tried to use MavLink (for OpenIPC) and had no luck as well... Just silence on the port.
I then tried the same set-up (same FTDI adapter, same model of flight controller), but flashed with iNav and MavLink works fine there

seriyps, 2024-09-19:
@catkira may you please reopen the issue?

seriyps, 2024-09-19:
UPD: I flashed iNav instead of BetaFlight on that drone and it produces telemetry.
[image]

hugepanic, 2025-03-13:
Same here, but I think the issue is worse than only MavLink   ---> I cant get ANY telemtry out of ANY UART! It is not limited to MavLink, as I have tested all other Protocols.

Method applied: Oszilloscope

Betaflight 4.5.1
F7 Flight controller -->  DAKEFPVF722

I have now switched to INAV and everything works as expected!

--> looks like a huge BUG to me.

englebert, 2025-03-16:
> Same here, but I think the issue is worse than only MavLink ---> I cant get ANY telemtry out of ANY UART! It is not limited to MavLink, as I have tested all other Protocols.
>
> Method applied: [...]

You need to add "TELEMETRY_MAVLINK" in the **CUSTOM DEFINES**  at the  **FIRMWARE FLASHER**

Mine works pretty good.

closed 2026-09-26 by blckmn