betaflight/betaflight #13489: battery indicator disappear 4.5 RC3

Gierzow, 2024-03-31:
### Describe the bug

When flying on HD zero build I approach battery super low voltage ignoring low battery warning than battery indicator dissapear along with the warning. From DVR looks like last battery voltage recorded was 19.5V for 6s pack 

### To Reproduce

Fly as hard as you can and drop the battery to low voltage ignoring low voltage warning

### Expected behavior

Battery indicator should stay on the OSD and low battery warning should be on 

### Support ID

```txt
a6cab286-a784-48a7-884f-85e8408b1686
```

### Flight controller

TMPACERF7

### Other components

HDZERO race v1 board and 90fps camera 
happymodel ep1 elrs 3.x

wterreb, 2024-04-06:
I also see that on some of my analog quads.   The battery voltage appear and disappear randomly on the OSD.

bilaal-s, 2024-05-10:
supporting evidence; timestamp included https://youtu.be/FbC6yjEG2iE?feature=shared&t=64

You will observe after a few seconds, the battery indicator and the warning indicator both disappear.

grepboy, 2024-11-24:
I have also observed this behavior using HD OSD, Brave F7 FC connected to DJI Vista with WTF OS.  Seems that as soon as any element starts to blink a warning, all of the OSD elements disappear.  When the warning element stops blinking, the whole OSD returns.

haslinghuis, 2024-11-24:
Either update video unit related firmware or in the mean while try

```bash
set displayport_msp_fonts = 0,0,0,0
save
```

closed 2025-08-10 by haslinghuis