betaflight/betaflight #13490: 4.5 RC3 - Disarmed when performing return to home

AdrianEddy, 2024-03-31:
### Describe the bug

Hi
I was just testing Return to Home on RXLOSS (activated via switch) today and the quad dropped from the sky (from 40m), when it returned to the GPS coordinates. The RTH procedure started fine, it flied back to the home coordinates at 40m altitude and then just disarmed with CRASH_PROTECTION

I tested Return to Home on this quad (and same config and BF version), but with different PIDs/Filters, and it worked fine first 2 times I tested

Today it just disarmed and dropped from the sky, without me touching anything (and it was a clear big open field, so it didn't crash into anything either).

The FC is T-Motor F7 HD with BMI270 gyro and I have GEP-M1025-DQ GPS, with Baro and Compass configured and detected in BF

I'm attaching blackbox log, hopefully it contains helpful information.

[btfl_001.zip](https://github.com/betaflight/betaflight/files/14816561/btfl_001.zip)

### To Reproduce

See the blackbox

### Expected behavior

I expected it to lower the altitude and land, instead of disarming at such high altitude

### Support ID

```txt
f7c98b29-bfbf-40cc-8365-9d0453931c9e
```

### Flight controller

T-Motor F7 HD

### Other components

GPS (GEP-M1025-DQ), TBS Nano RX, O3 Air Unit

ctzsnooze, 2024-04-01:
Sorry to hear about the crash.
Maybe don’t configure it to fly home so high until you’ve tested it out. Do lots of tests at low altitude until you are sure it is ok. 

Crash protection is enabled in case during the return you hit a tree or a person. It will disarm if the quad is not responding properly to the pid commands. The abrupt roll right should not have happened. Whatever caused that probably triggered the crash protection and caused the disarm. 

I don’t know what caused that, it doesn’t make much sense. We’ve had tens of thousands of rescues without this, and no change to that part of our code, so it’s not likely a bug. 

It’s possible that there was a motor desync or CPU overload but that is unlikely. 

Did you enable a Mag?  If you did, we switch to Mag for heading at this point and that could possible cause a need for an abrupt heading change if the Mag was not working or was giving very different values.  

You can make the return smoother by reducing P, D and maybe I. And there is a smoothing factor. You can read about that in the wiki.

If there are big steps during the return, the GPS may not be returning position values as expected. This can be due to a failure to respond to auto-configuration, or a marginal position fix. When the gps data changes abruptly the quad will also respond abruptly. But yes you can make it smoother. 

Be sure to not use 8k8k with rescue because the CPU load can be too much. Check in the status command in the CLI that the module is connected and configured. 

At this point that‘s all I can say, thanks for reporting. Please do more tests, close to takeoff and at low altitude over grass until you are sure it is reliable.

AdrianEddy, 2024-04-01:
Hmm, now that you mentioned Mag, yes I did have Mag enabled, and I did the calibration. However, Sometime between the calibration and that particular flight, I rotated the GPS mount 180degree, which also rotated the mag, but I didn't re-calibrate it. 
Does that sound like it may be the cause? The drone was flying fine until that point, is Mag used only in that GPS Rescue step?

If that's indeed the cause, that's clearly my error, but maybe there's something we can do to prevent such thing for other people?

I have 3.2 kHz PID loop due to BMI270

closed 2024-05-09 by github-actions[bot]