betaflight/betaflight #13491: Rolling blackbox storage

bvandevliet, 2024-04-01:
### Is your feature request related to a problem? Please describe

When storage is full, no new blackbox logs will be stored. I would like to have blackbox logging enabled at all times so that when something happens, I have a blackbox to analyze. But I don't want to check regularly if there is still sufficient free storage for new blackboxes to be recorded. 

### Describe the solution you'd like

When storage is full, begin overwriting existing blackboxes starting with the oldest (rolling method). 

### Describe alternatives you've considered

None. 

### Other information

Would be good to have this as an option we could turn on or off, so for anyone that prefers the way it works today, he/she would still be satisfied.

ledvinap, 2024-04-01:
Flash erase is slow (up to seconds). It would be necessary to buffer BB data during this time.

bvandevliet, 2024-04-01:
That's a valid point and perhaps buffering introduces additional drawbacks. Just thinking out loud here.. Let's say we reserve xx% of storage to be free at all times and have the erasing algorithm running at all times (so also when disarmed and not recording blackboxes). Could that work reliably?

ledvinap, 2024-04-02:
It will be possible, but considerable code changes would be necessary. IIRC BF firmware simply uses flash as one large block, always appending data at the end. 

BLACKBOX ERASE mode is available, but not that useful.

Any volunteer?

albert0ly, 2024-04-02:
Sorry to say, if you call it "blackbox" let it behave like "back box". Overwrite old records with new ones. No much use of stale records. Otherwise call it not "black box" but "log for turning"

ledvinap, 2024-04-02:
@albert0ly:  I'd gladly review your PR that will make "blackbox" behave like "back box"

haslinghuis, 2024-04-02:
@albert0ly We are happy for contribution on improving blackbox drivers

jawz101, 2024-05-22:
create a new file each 1MB, first in first out.  I don't know how the logging and flashing works though.  Is it deleting files or reformatting the chip?  I would imagine the latter would be a slower process.

> Sorry to say, if you call it "blackbox" let it behave like "black box". Overwrite old records with new ones. No much use of stale records. Otherwise call it not "black box" but "log for turning"

Either way, I would think a "log for tuning" would be a rolling storage just as a "black box" would.  If I'm going to fly my quad around and then analyze something I did, the thing I am interested in is probably in the last bit of flying before the logging stopped :/

bvandevliet, 2024-10-20:
Hi there, have there been any developments on this issue in the meantime? :)

Rambo-Sec, 2026-08-19:
Has there been any further discussion or technical exploration of this feature?

I recently encountered the exact problem described here. A flight controller with 16 MB of onboard flash recorded approximately 205 seconds at a 1/4 sampling rate. During a later flight, the storage became full about 12 seconds after arming. The aircraft subsequently experienced a sudden loss of control, but the decisive portion was not recorded because Blackbox had preserved the earliest flights instead of the most recent data.

For failure analysis, rolling storage seems like the behavior most users would expect from something called a blackbox. Ideally, an optional mode could continuously replace the oldest complete logs or storage blocks while preserving the newest flights.

I understand that implementing this safely on SPI flash may involve erase block management, flash wear, metadata integrity during power loss, and changes to FlashFS. I also appreciate that Betaflight is maintained by volunteers, so I am not requesting a timeline or commitment.

Could someone familiar with the Blackbox or FlashFS implementation clarify what the primary obstacle is? Is this mainly a software architecture issue, a hardware limitation, or simply work that has not yet had a contributor? Any guidance on the components that would need to change would be helpful for anyone interested in exploring a future implementation.

nerdCopter, 2026-08-19:
https://github.com/betaflight/betaflight/pull/15197