betaflight/betaflight #13504: OMNIBUSF4SD does not work with IBUS in BF 4.3.5 RC 3

wterreb, 2024-04-06:
### Describe the bug

IBUS receiver on my  OMNIBUSF4SD works perfectly in BF  4.3.0.     
But after I upgrade to 4.3.5 RC3 it does not work anymore. 
If I go back to 4.3.0 again then it works again.

### To Reproduce

Upgrade OMNIBUSF4SD to BF 4.3.5 RC3 and choose IBUS as the radio protocol.

### Expected behavior

After upgrade, IBUS receiver should continue to be detected and work

### Support ID

```txt
00f2b709-6ece-48cc-9420-eec874dacae1
```

### Flight controller

OMNIBUSF4SD

### Other components

FlySky2A IBUS receiver

### How are the different components wired up (including port information)

Receiver connected to Ibus input

### Add any other context about the problem that you think might be relevant here

No

blckmn, 2024-05-12:
@SteveCEvans can this be in maintenance also please.

SteveCEvans, 2024-05-12:
I don't have any IBUS kit. This is a classic case for a `git bisect` to find the commit that fixed it. Do any of the devs use this?

closed 2024-06-19 by github-actions[bot]