betaflight/betaflight #13505: qmc5883 & DSP310 I2C bus issue: no data when enable baro & mag

lida2003, 2024-04-06:
### Describe the bug

It is something like [Mag and Baro will not work at the same time, even disarmed state](https://github.com/betaflight/betaflight/issues/13360)

It was supposed to fix at [ Fix qmc5883l lockup #13467 ](https://github.com/betaflight/betaflight/pull/13467).

I have tried latest version, git 54570328388190eb48900e0db6f2f271192019a7

If disable mag, then baro is OK.

### To Reproduce

1) Flash betaflight git 54570328388190eb48900e0db6f2f271192019a7
2) Enable baro & mag
3) There is no data in sensor tab, but I2C detects DPS310 & qmc5883 chip
4) Disable mag, reboot
5) We get data in sensor tab.

### Expected behavior

As always, we can get baro & mag data at disarm/arm state.

### Support ID

```txt
74beadf5-38b3-4480-ba89-c9a80b0a4bcc
```

### Flight controller

AT32

haslinghuis, 2024-04-06:
- duplicate of #13360

Giammi, 2024-04-06:
@lida2003 please test #13509 it worked for me.

https://github.com/betaflight/betaflight/issues/13360?notification_referrer_id=NT_kwDOA_EuS7M5NDMxMDE5MjUxOjY2MTM3Njc1#issuecomment-2041169611

lida2003, 2024-04-07:
> @lida2003 please test #13509 it worked for me.

#13509 tested.
a) both mag & baro enabled:   no data output
b) mag disabled & baro enabled: baro data is OK
c) mag enabled & baro disabled: no data output (always 0)

closed 2024-04-09 by haslinghuis