betaflight/betaflight #13510: Yaw twitch on rappid stick centering with ezlanding active and low throttle

tbolin, 2024-04-06:
### Describe the bug

The yaw might twitch when rapidly centering the stick with ezlanding on an low throttle.

[image]

[ez_land_yaw_twitch.zip](https://github.com/betaflight/betaflight/files/14894987/ez_land_yaw_twitch.zip)

I have a pretty good idea what is going on here and will start working on a fix.
The upper limit decreases extremely quickly (in the screenshot above from max to min in about 1/50 of a second), causing the motors to slow down before the the yaw motion have stopped. The change in momentum of the motors then causes the yaw rate of the quad to increase instead, before slowly slowing down again. 

### To Reproduce

Turn on ezlanding and ideally use a quad that has poor yaw authority or turn up the max yaw rate
Make a sharp pure yaw motion with the throttle at zero and center the stick as quickly as possible.
The yaw rate will stop going towards zero or even twitch.

### Expected behavior

No yaw twitching

### Support ID

```txt
NA
```

### Flight controller

Any

Mick51, 2024-04-07:
I have the same problem with EZLANDING set to Legacy and therefore turned off. I didn't have this problem before switching to RC3.
FC Speedebee F405 V3
Betaflight 4.5 RC3

tbolin, 2024-04-07:
@Mick51 
> I have the same problem with EZLANDING set to Legacy and therefore turned off. I didn't have this problem before switching to RC3.
> FC Speedebee F405 V3
> Betaflight 4.5 RC3

Could you please post a log?

sugaarK, 2024-04-15:
@tbolin  are you on our discord

Mick51, 2024-04-15:
> @tbolin are you on our discord

Problem solved by flashing the latest dev version.  😉

tbolin, 2024-04-15:
@sugaarK 
> @tbolin are you on our discord

Yes, the nick is Zoggbarr

closed 2024-05-23 by github-actions[bot]