betaflight/betaflight #13515: Beeper not working for magnetometer calibration sequence

virusapex, 2024-04-09:
### Describe the bug

When performing _stick commands_ or initiating magnetometer calibration from configurator, beeper isn't making any sounds. It's working for accelerometer calibration sequence, making a single beep for both options.

### To Reproduce

Pressing "Calibrate Magnetometer" in Betaflight Configurator or using _stick commands_ from the transmitter.

### Expected behavior

Hearing two short beeps confirming that I initiated the calibration process.

### Support ID

```txt
9d6057b2-442e-46c4-a9e3-b064ffbcae02
```

### Flight controller

GEPRCF722_BT_HD_V2

### Other components

GEP-M10-DQ (M10 GPS, QMC5883L Magnetometer, DPS310 Barometer)

### How are the different components wired up (including port information)

All through UART5

Giammi, 2024-04-22:
You have to activate ACC_CALIBRATION and GYRO_CALIBRATED in the beeper section (https://github.com/betaflight/betaflight/pull/13487#issuecomment-2040676977)

haslinghuis, 2024-05-12:
Closing as configuration issue

closed 2024-05-12 by haslinghuis