betaflight/betaflight #13528: ClRacingF7 + Crossfire = No Telemetry

CjlaBuk, 2024-04-15:
I'm back to the hobby after a couple of years and after updating my FC to 4.5 I'm getting no telemetry from Betaflight on my radio.
My previous firmware was 4.2 and it used to work fine as i remember from like 4 years ago.
I do get stick inputs in the configurator and all the telemetry coming from the nanoRX (like lq, tpwr) but no battery voltage/current draw or anything from BF
I did check my cables few times already and they are ok, also tried changing UARTS and making them inverted/halfduplex in CLI

Support ID: 393476b8-e4f9-4a85-9935-980c19b453b7
Help!

_Originally posted by @CjlaBuk in https://github.com/betaflight/betaflight/discussions/13523_

haslinghuis, 2024-04-15:
- What is the firmware version and settings used on CRSF?
- Please provide tasks output (two times in a row)
- Disable ADC on the radio
- If enabled then disable dynamic power
- RX rate: 150 seems very high (have to check)
- set crsf_use_negotiated_baud = OFF (have to look that up)
- JEDEC ID=0x00efaa21(FLASH_W25N01G) (can you disable this temporary)
- Try M10 GNNS module on 57600 baud
- Try lowering DSHOT600 to DSHOT300

Please consider using our [Discord Server](https://discord.com/invite/gV4XSq3fus) for support.

CjlaBuk, 2024-04-15:
> * What is the firmware version and settings used on CRSF?
6.19
> * Please provide tasks output (two times in a row)

# 

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)      9       2       0    0.0%    0.0%         0      0     55       1
01 - (         SYSTEM)    995       5       0    0.4%    0.0%         2      3   5245       1
02 - (           GYRO)   7949      14       2   11.1%    1.7%        64      0  41940       0
03 - (         FILTER)   3976      38      18   15.1%    7.3%       425      0  20970       0
04 - (            PID)   3976      72      42   28.6%   16.9%       979      0  20970       0
05 - (            ACC)    996      10       2    0.9%    0.2%        14      2   5243       5
06 - (       ATTITUDE)     99      19       9    0.1%    0.0%         5      2    528      13
07 - (             RX)    143      29      13    0.4%    0.1%        32      8   2313      20
08 - (         SERIAL)     99  341287       2 3378.7%    0.0%       598      0    515      58
09 - (       DISPATCH)    994      10       0    0.9%    0.0%         4      2   5244       2
10 - (BATTERY_VOLTAGE)    199      12       2    0.2%    0.0%         2      1   1051       8
11 - (BATTERY_CURRENT)     50       4       0    0.0%    0.0%         0      0    266       3
12 - ( BATTERY_ALERTS)      4       4       2    0.0%    0.0%         0      0     28       3
13 - (         BEEPER)     99       8       0    0.0%    0.0%         0      0    514       6
14 - (            GPS)    152      55      11    0.8%    0.1%         4     36    607      29
15 - (     GPS_RESCUE)     99      14       5    0.1%    0.0%         3      2    528      13
17 - (       ALTITUDE)     99       7       2    0.0%    0.0%         1      0    514       6
18 - (      TELEMETRY)    498      17       0    0.8%    0.0%         5      2   2587       5
20 - (            OSD)      9      34      11    0.0%    0.0%         4      6    648      12
22 - (            CMS)     18       6       1    0.0%    0.0%         0      0    107       5
24 - (    ADCINTERNAL)      2       4       1    0.0%    0.0%         0      0      6       3
25 - (       PINIOBOX)     18       7       2    0.0%    0.0%         0      1    107       6
26 - (SPEED_NEGOTIATION)     99      14       1    0.1%    0.0%         0      1    514      10
RX Check Function                  12       1                         3
Total (excluding SERIAL)                                26.3%

# 
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       6       0    0.0%    0.0%         0      0    443       3
01 - (         SYSTEM)    995      10       0    0.9%    0.0%        26     19  44098       1
02 - (           GYRO)   7955      13       2   10.3%    1.8%       786      0 352785       0
03 - (         FILTER)   3978      32      19   12.7%    7.5%      3699      0 176392       0
04 - (            PID)   3976      67      44   26.6%   17.6%      8597      0 176393       0
05 - (            ACC)    996      12       2    1.1%    0.2%       130     27  44098       8
06 - (       ATTITUDE)     99      20       9    0.1%    0.0%        49      2   4410      14
07 - (             RX)    150      22      13    0.3%    0.1%       306      8  19752      17
08 - (         SERIAL)     99    5448       1   53.9%    0.0%       612      0   4408      30
09 - (       DISPATCH)    994      10       0    0.9%    0.0%        38     15  44099       1
10 - (BATTERY_VOLTAGE)    199      12       2    0.2%    0.0%        22      5   8820      11
11 - (BATTERY_CURRENT)     50       5       0    0.0%    0.0%         2      0   2218       3
12 - ( BATTERY_ALERTS)      5       7       2    0.0%    0.0%         0      0    222       5
13 - (         BEEPER)     99      11       0    0.1%    0.0%         4      2   4408       2
14 - (            GPS)    166      33      13    0.5%    0.2%        73      2   6100      29
15 - (     GPS_RESCUE)     99      15       5    0.1%    0.0%        27      2   4410       8
17 - (       ALTITUDE)     99      11       2    0.1%    0.0%        12      2   4408       5
18 - (      TELEMETRY)    497      10       0    0.4%    0.0%        16     15  22040       1
20 - (            OSD)     12      40      11    0.0%    0.0%        82      3   7786      10
22 - (            CMS)     20      10       1    0.0%    0.0%         1      1    887       7
24 - (    ADCINTERNAL)      1       7       3    0.0%    0.0%         0      0     45       6
25 - (       PINIOBOX)     20      13       2    0.0%    0.0%         2      1    887      11
26 - (SPEED_NEGOTIATION)     99      12       2    0.1%    0.0%         8      0   4408       9
RX Check Function                  16       1                        36
Total (excluding SERIAL)                                27.4%

> * Disable ADC on the radio
√ Was disabled

> * If enabled then disable dynamic power
√ Done. Nothing changed

> * RX rate: 150 seems very high (have to check)
√ Changed to 50 got 1 more sensor "ANT"

> * set crsf_use_negotiated_baud = OFF (have to look that up)
√ Tested with ON, still nothing

> * JEDEC ID=0x00efaa21(FLASH_W25N01G) (can you disable this temporary)
???? How do i do that ????

> * Try M10 GNNS module on 57600 baud
√ Done

> * Try lowering DSHOT600 to DSHOT300
√ Done
> 
> Please consider using our [Discord Server](https://discord.com/invite/gV4XSq3fus) for support.

CjlaBuk, 2024-04-15:
Re-soldered RX/TX cables to pads 3 and 4 in the receiver. Configured them as rx & tx, now everything works.

closed 2024-04-15 by CjlaBuk