betaflight/betaflight #13538: EzLanding: better landing detection

KondratokGK, 2024-04-17:
### Is your feature request related to a problem? Please describe

EzLanding active when drone in free fall.

### Describe the solution you'd like

Add additional condition for EzLanding: Enable EzLanding only when G-force is from 0.5G to 1.5G directed down (Like arm angle).

### Describe alternatives you've considered

EzLanding AUX switch, but it`s sounds really stupid.

themeadery, 2024-04-29:
I had some weird issues a few times flying the new EZLANDING mixer type. Someone suggested putting it on a switch so it's only active when you come in to land. So yeah I think having it on an AUX switch is a good idea, but I looked at how to implement that and I can't find a way.

tbolin, 2024-06-27:
@KondratokGK 

If you have a GPS EzLanding will already disabled over a certain speed (default 5 ms/s, adjustable through `ez_landing_speed` in cli). If you don't have a GPS and are having trouble with instability during drops I suggest increasing `ez_landing_limit` a bit.

Inactivating EzLanding when the acc magnitude is not close to 1g, would pretty much cause it to be disabled during any landing, especially rough ones. Rough landings can also saturate the gyro causing the attitude estimate to be wrong. Besides there have been more than one instance when I was very happy that EzLanding was active with the quad upside down. 

@themeadery can you describe some of these weird issues?

closed 2024-08-04 by github-actions[bot]