betaflight/betaflight #13539: Softserial SmartPort not working in 4.5-RC4

psfales, 2024-04-17:
### Describe the bug

I'm using a Diatone Mamba F405 Mk2 (target FURYF4OSD) with a Frsky RXS4 and S.Port telmetry on Softserial 1.   In 4.4.3, it works fine.   But with 4.5-RC4 there is no telemetry output.  I have SMARTPORT as the telemetry protocol in the cloud build,  Smartport enable for SOFTSERIAL 1 on the ports tab, and telemetry turned on in the receiver tab.  (Same for both 4.4.3 and 4.5)  In 4.5, I had to change from SERIAL_TX 11 to SOFTSERIAL_TX 1.  But in 4.5 there is no telemetry and (using an oscilloscope), no smartport data flowing on that wire. 

### To Reproduce

RESOURCE SOFTSERIAL_TX 1 C09
- Enable Smartport telemetry on Softserial 1 on the ports tab
- Enable telemetry on the receiver tab

No telemetry transmitted on pin C09 (labelled PPM1 on this FC)

### Expected behavior

I/O is limited on this FC, and the PPM pin is adjacent to the other receiver connections, so I really would like to use softserial for this board.   But for now, 

### Support ID

```txt
Id: "96420015-4a6f-4439-b7be-b9fca941ca67"
```

### Flight controller

Diatone Mamba F405 Mk2 from getfpv

### Other components

Frsky RXSR,  Betian BN880 GPS, no VTX

### How are the different components wired up (including port information)

Receiver is SBUS on RX1.   Receiver telemetry is Smartport on the "PPM" pin (C09) using Softserial

haslinghuis, 2024-04-17:
Please test RC3, RC2, RC1. 
If it starts working here - we can narrow down to an individual PR.

psfales, 2024-04-17:
Good idea - I should have thought of that.  Unfortunately, however, it does not work even in RC1

I don't know if the support data is useful, but if it helps, it is 
Id: "0456a472-dd2b-44e8-aa8d-d256d233a847"

haslinghuis, 2024-04-17:
https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC1

psfales, 2024-04-17:
I don't understand the significance of that last comment with the link to RC1.  Is there something more that I need to test on RC1?  I didn't think it was necessary to test RC2 or RC3 since the problem is happening for me on both RC1 and RC4.

haslinghuis, 2024-04-17:
Sorry just left as a note to have easy access to the list of PR's for reference.

From here we can check early firmware: https://github.com/betaflight/betaflight-nightlies/releases using local builds and use a 4.5 dump to restore configuration.

haslinghuis, 2024-04-17:
Can you try with RC4:

```bash
timer C09 NONE
dma pin C09 NONE
save
```

psfales, 2024-04-17:
Unfortunately, that didn't make any difference
Id: "8de2787f-3d55-4185-bc91-50a7956adc9d"

psfales, 2024-04-17:
Not sure if this is of any use at all, but in 4.4.3 where it is working, I have the softserial (Smartport) baud rate set to Auto.    I am using "Auto" in 4.5 as well.   I found a web page (https://code.google.com/archive/p/telemetry-convert/wikis/FrSkySPortProtocol.wiki) which says that Smartport runs at 57600, so I tried setting it to that.   It still doesn't work, but I noticed that whenever I change it to 57600, something keeps setting it back to 19200 - "auto" will "stick", but 57600 does not. 

I realize it's a long shot, but wondering if the "Auto" setting is trying to the same thing... Setting it to 57600, but unable to make it "stick"

haslinghuis, 2024-04-17:
Baudrate for softserial is restricted to 19K2

haslinghuis, 2024-04-17:
What is the baudrate used on 4.4.3 ?

haslinghuis, 2024-04-17:
@SteveCEvans :point_up:

haslinghuis, 2024-04-17:
@psfales can you check the PR?

psfales, 2024-04-17:
The code change looks very promising, but I'm probably doing something wrong, as I've never tested a PR before.   I did these steps:
- Select "Development" releases
- Select "4.5-dev"
- In the box at the lower left, there were a bunch of choices, but PR #13540 wasn't listed in the dropdown, so I just type "#13540"
- It did a new cloud build (unlike earlier cases where the build was cached)

But it didn't work.   When I set baud rate to 57600, and reboot, it gets re
set to 19200

haslinghuis, 2024-04-17:
Have to paste in `#13540` in select pull requests or commit field

psfales, 2024-04-17:
That's what I tried to do, but it doesn't seem to "take."   I think what was happening is that when I tried to type in something that wasn't in the dropdown list, it just goes to the first thing on the list which which is the "Latest (master)"

Do I need a different version of the configurator?    Do I have do do something to override the dropdown list?

haslinghuis, 2024-04-17:
Crafted from your last support ID:

[betaflight_4.5.0_STM32F405_FURYF4OSD.zip](https://github.com/betaflight/betaflight/files/15017012/betaflight_4.5.0_STM32F405_FURYF4OSD.zip)

haslinghuis, 2024-04-17:
Configurator 10.10RC4 or nightly

psfales, 2024-04-17:
Well, one step closer, but not there yet.   I was using 10.10-RC4, but it would not let me type in the hashtag-PR until I exited and restarted the configurator.   THEN it let me type in the PR number, and did a cloud build.

The behavior is different now:  I can set the softserial baud rate to 57600 and have it "stick" across a reboot instead of getting set back to 19200.

But, sadly, the telemetry behavior is unchanged - I'm not getting Smartport telemetry

haslinghuis, 2024-04-17:
Hmmm too bad.

psfales, 2024-04-17:
FWIW, in 4.4.3, it doesn't matter whether I set the baud rate to 19200, 57600, or Auto - the value "sticks" and the telemetry works.    That makes me think that "Smartport" is ignoring the value specified on the ports page and just using whatever is actually needed by Smartport.     Is there a chance that Smartport in 4.5 is trying to set the baud rate (or some other aspect of the serial port configuration) and not being allowed to do so?

Even though Smartport normally runs over a TX pin, I think it does need to see data coming back from the receiver (like timing pulses).   Any possibilities there?

ledvinap, 2024-04-17:
@psfales: Smartport is 1-wire, single direction (TX only) and inverted. There is no feedback from receiver.

Can you capture data using Saleae/logic analyzer/oscilloscope?

psfales, 2024-04-17:
According to https://code.google.com/archive/p/telemetry-convert/wikis/FrSkySPortProtocol.wiki the protocol is "single wire, meaning that both tansmit [sic] and receive are using the same wire, so they cannot talk at the same time (half duplex)"

That's consistent with what I see on the logic anaylzer.   If I disconnect the Smartport wire, and monitor the receiver side, I see a short burst every 12msec or so.

I've attached two pictures, one from 4.4.3 and one from 4.5.    
[image]
[image]

In the 4.4.3 image, I believe the short burst at 0, 12, and 24 msec is from the receiver, and the longer burst at 4 and 25msec is from the FC.

In the 4.5 image, we still get the timing data from the receiver, but nothing is sent by the FC.

psfales, 2024-04-17:
For the record, I have tlm_halfduplex=ON in both 4.4.3 and 4.5, and I have experimented with both tlm_inverted=ON and tlm_inverted=OFF.   In 4.4.3, both work (leading me to think that smartport is just doing it's own thing and ignoring the setting), but both ON and OFF do not work in 4.5.

psfales, 2024-04-17:
Based on the information on that protocol wiki page, I believe that the FC should be seeing sync bytes (0x7E, followed by a sensor ID) about every 12 msec.   Are there any debugging tools that would tell me if the FC is seeing those bytes?
 
 If there is such a thing, I could run the test in both 4.4.3 and 4.5 and see if they are consistent.

psfales, 2024-04-18:
In 4.4.3, this works...   I can use "serialpassthrough 30 57600" to monitor the softserial port.  It prints one line about every half-second:
[image]

In 4.5, the serialpassthrough does not work (is this a bug?) even though port 30 is listed in the "serial" command:
[image]

psfales, 2024-04-18:
Still no luck.   When I did that change, I got a warning, but it apparently accepted it.  Howe

```
# resource softserial_rx 1 C09

NOTE: C09 already assigned to SOFTSERIAL_TX 1.

Resource is set to C09
```

And I'm still unable to do a passthrough as I could in 4.4.3:

```
# serialpassthrough 30
Port1: 30 
Port2: 20 
Port1 could not be opened.
```

I tried baud rates of 19200, 57600, and AUTO (all of which worked in 4.4.3) with the same (lack of) results in 4.5.  I was able to use the logici analyzer that the bit time of the sync bytes coming from the receiver is about 17 miciroseconds which is consisent with a baud rate of 57600.

psfales, 2024-04-18:
I also tried disabling the telemetry on softserial 1 on the ports tab.  In theory, softserial 1 should be completely unused (free) now.   I stlll get "Port1 could no be opened" when I do "serialpassthrough 30"

psfales, 2024-04-18:
And I tried setting "resource softserial_tx_1 none" while leaving "resource softserial_rx 1 c09"

I don't expect that to work for telemetry, but if serialpassthrough worked, I should be able to see the sync bytes coming from the receiver.    I still get "Port1 could no be opened"

ledvinap, 2024-04-18:
Strange thing is that `could not be opened.` happens only when port is not open. But timer is claimed (TIM8: CH4 : SOFTSERIAL_TX 1) and there is little possibilities to fail after that.

Can you try `resource show` ?

Also, try fresh supportId, with #13540 version.

psfales, 2024-04-18:
"resource show" is below.    The SupportID is Id: "e8fbcc52-c07e-4f85-9895-9d2bc13ea90e"

```
# resource show
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_SDI 1
A07: SPI_SDO 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: PINIO 1
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_SDI 2
B15: SPI_SDO 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: USB_DETECT
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_SDI 3
C12: SPI_SDO 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

```

psfales, 2024-04-18:
"Strange thing is that could not be opened. "    Can anyone else reproduce that?   (Seems like it should be possible to grab any spare pin, assign a softserial_tx  and/or softserial_rx to it, and see if serialpassthrough works.)    If not, maybe there is a bug there??

haslinghuis, 2024-04-18:
@ledvinap After reverting https://github.com/betaflight/betaflight/commit/42267349db0c92a0e1063bd1d07633a3322dfb07 it seems to work:

```
# serialpassthrough 30
Port1: 30 
Port2: 20 
Port1 is already open, baud = 57600.
Port1 baud rate change over USB enabled.
Forwarding, power cycle to exit.
```

ledvinap, 2024-04-18:
@psfales : 

Can you try `timer C9 AF3` ?

And also try starting with default config (don't restore saved one)

@haslinghuis:
even with https://github.com/betaflight/betaflight/commit/42267349db0c92a0e1063bd1d07633a3322dfb07 19200 shall work. And https://github.com/betaflight/betaflight/pull/13540 increased the limit to 57600

psfales, 2024-04-18:
`timer c9 af3` did not help

I tried with a blank configuration.   To my surprise, "30" is not listed in the output of the "serial" command:

```
# resource escserial none
Resource is freed

# resource SOFTSERIAL_TX 1 C09

Resource is set to C09

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# serialpassthrough 30
Port1: 30 
Port2: 20 
Port1 could not be opened.

# resource escserial none
Resource is freed

# resource SOFTSERIAL_TX 1 C09

Resource is set to C09

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# serialpassthrough 30
Port1: 30 
Port2: 20 
Port1 could not be opened.
```

ledvinap, 2024-04-18:
Also, after attempting `serialpassthrough`, get `timer show`

psfales, 2024-04-18:
I don't see how to do `timer show` because after `serialpassthrough`, the cli is locked out until power cycle.

ledvinap, 2024-04-18:
@pfales: 

```
feature SOFTSERIAL
# pin must be assigned
resource SOFTSERIAL_TX 1 C09  
resource SOFTSERIAL_RX 1 C09
# try without this first
timer C09 AF3
save

ledvinap, 2024-04-18:
You should stay in `cli` if `serialpassthrough` fails

psfales, 2024-04-18:
```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# serialpassthrough 30
Port1: 30 
Port2: 20 
Port1 could not be opened.

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 1
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 3
TIM4: FREE
TIM5:
    CH1 : LED_STRIP
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1 : CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

haslinghuis, 2024-04-18:
Updated #13540

Result:

```bash
# serialpassthrough 30
Port1: 30 
Port2: 20 
Port1 is already open, baud = 19200.
Port1 baud rate change over USB enabled.
Forwarding, power cycle to exit.
```

ledvinap, 2024-04-18:
@haslinghuis : That's with smartport on softserial?

haslinghuis, 2024-04-18:
Bare board as I don't have frsky - but on master port did not open.

haslinghuis, 2024-04-18:
@psfales please try with updated `#13540`

psfales, 2024-04-18:
Still no luck.  I rebuilt with the PR and it looked like it is doing the right thing.  (I saved a copy of the build log if you want see it, or have me check anything, but it included 
```
VERSION:   4.5.0-dev
COMMIT:    pull/13540/head
CONTAINER: master
BOARD:     FURYF4OSD/STM32F405
STARTED:   2024-04-18 18:06:20Z
```

There is no telemetry, and I serialpassthrough can't open the port.    Since the serialpassthrough failed, I'm able to run `timer show` which gives:# timer show

```
Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 1
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 3
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1 : CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FRE

```

haslinghuis, 2024-04-18:
WTF, local build works - cloud build does not :exploding_head: 

[betaflight_4.5.0_STM32F405_FURYF4OSD.zip](https://github.com/betaflight/betaflight/files/15028979/betaflight_4.5.0_STM32F405_FURYF4OSD.zip)

psfales, 2024-04-18:
Progress!   I need to reproduce this again, and then start stepping through the backup file, but I loaded the local build file and then in the **clean** configuration did

```
**feature softserial**
resource escserial none
resource softserial_tx 1 c09
```

**AND I GET TELEMETRY**

But when I load my stored backup file, the telemetry stops.   I'll see if I can figure out what is doing that!

ledvinap, 2024-04-18:
post `diff` of your backup ...

ledvinap, 2024-04-18:
I can confirm that smartport over softserial is 'working' when 19200 limit is disabled and smartport pin properly configured. I did not test CPU load yet. As stated above, it may cause issues in other code.

psfales, 2024-04-18:
So, this is wierd.   I start with a clean configuration, and run

```
feature SOFTSERIAL
feature TELEMETRY
resource ESCSERIAL 1 NONE
resource SOFTSERIAL_TX 1 C09

```

Telemetry works.    Then when I load my backup file, the telemetry stops!    However, all four of those lines are in my backup file, just in a different order.   So, it's either the order of the commands, or there is something else in the backup file that is interfering.   Here is the backup file that "kills" it:
[BTFL_cli_backup_quad2_RC4_20240417_122711_FURYF4OSD.txt](https://github.com/betaflight/betaflight/files/15029430/BTFL_cli_backup_quad2_RC4_20240417_122711_FURYF4OSD.txt)

haslinghuis, 2024-04-18:
We still have a big issue as it works with local build and not with cloud build - using same key.

psfales, 2024-04-18:
SUCCESS!    The problem was that I had `set tlm_inverted=ON` in my backup file.   

With that ONE line removed (and no need to re-order the resource lines) my backup file works, I have telemetry, AND I have serial passthrough.

That was with the local build file.  Going to try the cloud build now...

psfales, 2024-04-18:
I'm seeing the same problem.  Telemetry and serialpassthrough work with the .zip file that @haslinghuis sent a short while ago (and with the bad line removed from the backup file), but the one from the cloud build does not work.

psfales, 2024-04-18:
Actually, the cloud build DOES work for me.   (I thought it was not working at first, but I think I forgot include the PR in the cloud build.   I redid the cloud build, and that is working now)

haslinghuis, 2024-04-18:
Weird - had checked twice locally. But checking again it now works :zany_face:

psfales, 2024-04-18:
This is great!   Thanks for all your help!

psfales, 2024-04-23:
I looked over the PR, and I'm a bit confused.   Will this "just work" in 4.5 (as it did in 4.4.3)?  Or do I need to do some sort of "override" on my end to enable it?

psfales, 2024-05-20:
For the record, I checked the CPU usage on my f405. It was about 49% without softserial and 55% with softserial running smart port at 57600. So, while 6% overhead is not insignificant, I don't think it's a showstopper either. I'm glad I was able to get this working in 4.5. thanks for all the help!

PR #13540 (2024-04-23): Fix smartport over softserial using override
- Fixes #13539
- Using constrain does not permit serialpassthrough
- SmartPort works only with 57600
- GPS and blackbox won't really work with 19K2 baud. See #13303
- See also #13305

How to test:
- Need to add `OVERRIDE_SOFTSERIAL_BAUDRATE` custom define to override the baudrate limit.

[image]

**DISCLAIMER:** _**THE USE OF THIS DEFINE IS UNSUPPORTED**_