betaflight/betaflight #13553: GPS stops working after plugging in the battery

dawidgora, 2024-04-19:
### Describe the bug

Hey, when I plug my drone into my computer and connect it to the betaflight configurator, I can see that the GPS is working fine. I see a red/yellow icon on top and `GPS: connected, UART3 115200 (set to 115200), configured, version =  M10` when typing `status` in the CLI. 

Once I plug in the battery to the drone GPS on top goes gray. `status` returns me: `GPS: connected, UART3 115200 (set to 115200), NOT CONFIGURED, version =  M10`. What could be the reason?

### To Reproduce

Description already describes steps to reproduce.

### Expected behavior

GPS works fine.

### Support ID

```txt
###

# # Problem description

# #

# ###

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Apr 15 2024 / 14:10:20 (7c8a55258) MSP API: 1.46
# config rev: 4cf8830
# board: manufacturer_id: BEFH, board_name: BETAFPVF722

# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=49degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3786, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MSP (53 x 20)
BUILD KEY: 7ad81a10e3614d2fead106d0747dd8c6 (4.5.0-RC4)
System Uptime: 33 seconds, Current Time: 2024-04-19T16:10:33.277+00:00
CPU:52%, cycle time: 127, GYRO rate: 7874, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - INIT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT CONNECTED, UART3 19200 (set to 115200), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS THROTTLE CLI MSP RESCUE_SW DSHOT_TELEM

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=83968

# dump master

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Apr 15 2024 / 14:10:20 (7c8a55258) MSP API: 1.46
# config rev: 4cf8830

# start the command batch
batch start

board_name BETAFPVF722
manufacturer_id BEFH

# name: Pavo25 V2

# resources
resource BEEPER 1 C14
resource MOTOR 1 B00
[... 1061 lines cut ...]
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

#
```

### Flight controller

BETAFPVF722

### Other components

PAVO25 V2
ELRS
DJI O3
GPS FlyFishRC M10 Mini
VIFLY GPS Mate

### How are the different components wired up (including port information)

[image]

dawidgora, 2024-04-19:
Ok, I just removed the SBUS wire from my O3 connection and now it's working.

closed 2024-04-19 by dawidgora

Turnip70, 2025-02-13:
Hej Dawid, Krzysiek z tej strony mam ten sam nierozwiązany problem, czy mogę się z Tobą skontaktować ?

dawidgora, 2025-02-14:
Cześć @Turnip70, problemem było to, o czym pisałęm w swoim poprzednim komentarzu. Wyciągnąłem przewód SBUS z kabla łączącego O3 z FC i działało. Innym rozwiązaniem jest podpięcie GPS do innego UART. Możesz się ze mną skontaktować, ale nie chcę tutaj umieszczać dokładnych namiarów. Możesz spróbować mnie znaleźć, np na Linked In, albo podrzucić swoje namiary to ja się do Ciebie odezwę.

Turnip70, 2025-02-14:
Krzysztof Rzepniewski :) co prawda problem rozwiązałem, ale możemy wymienić się doświadczeniami u mnie problem polegał na firmware ;)

dawidgora, 2025-02-14:
https://www.linkedin.com/in/dawidgora/