betaflight/betaflight #13556: HMC 5883 Magnetometer not working in 4.5.x anymore

SteffenFPV, 2024-04-20:
### Describe the bug

Magnetometers of type HMC5883 are correct detected in status, but no raw data in sensors tab. It is not a question of I2C settings or calibration. 

### To Reproduce

Enable Magnetometer in Firmware flasher. Flash BF 4.5.x. Connect HMC5883 sensor. Enable Magnetometer in Configurator. Check Sensors Tab and rotate Magnetometer.

### Expected behavior

HMC5883 works with same function as QMC5883 on BF 4.5.x.

### Support ID

```txt
6e55fd59-0dea-4ed7-af99-a8f1f3d62727
```

### Flight controller

Matek F722 SE

### Other components

Beitian BN-880 GPS/Mag

### Add any other context about the problem that you think might be relevant here

Magnetometers of type QMC5883 working properly. 
BF up to 4.4.x HMC and QMC sensors working correct.

haslinghuis, 2024-04-20:
Please test  #13537

Duplicate of #13338

haslinghuis, 2024-04-20:
Duplicate of #13338

wterreb, 2024-04-26:
Sorry to chime it on another ticket.   Not sure its related but no matter how much I tried I could also not manage to get BF to even detect my my GoKu GM10 Compass (MC5883L)

ledvinap, 2024-04-26:
@wterreb: Please open new issue. This one is duplicate, probably fixed by https://github.com/betaflight/betaflight/pull/13537

haslinghuis, 2024-04-26:
Fixed in #13537 :crossed_fingers:

closed 2024-04-26 by haslinghuis