betaflight/betaflight #13557: OSD says "Emuflight"

enterobotica, 2024-04-20:
### Describe the bug

I installed Emuflight on my 3.5" cinewhoop to test and went back to Betaflight. Now the FC is running Betaflight 4.5.0 RC4 but the OSD still says Emuflight in welcome message when powering up and the OSD is not taking any input from BF configurator 

### To Reproduce

FC: HGLRCF405V2

Install Emuflight > Install Betaflight with full chip erase > OSD still showing Emuflight

### Expected behavior

OSD should say the correct firmware version and show the right setup with respect to selected parameters to show, along with placement and typeface/font

### Support ID

```txt
c2152b00-d721-4293-98c5-6ae822adf176
```

### Flight controller

HGLRC SPECTER F460 STACK (Target: HGLRCF405V2)

### Other components

RX Protocol: iBus
VTX: Cyclone XF5804 Pro 5.8GHz 

### How are the different components wired up (including port information)

Motors and RX are soldered to the ESC and FC respectively, everything else is connecting via JST connectors
GPS, Magnetometer, LED are also installed via JST connectors

### Add any other context about the problem that you think might be relevant here

Used betaflight for a couple of weeks
Requested for addition of my target from Emuflight
Installed Emuflight, got all basic parameters to work including OSD
Installed Betaflight again after changing the way the stack was mounted (added bolts at bottom to try to reduce noise in YAW)
OSD now was showing Emuflight instead of Betaflight in welcome message.
Uninstalled Emuflight configurator, betaflight configurator, restarted PC
Installed Betaflight, tried hard resetting board and installing BF, still shows EmuFlight on OSD

haslinghuis, 2024-04-20:
Use OSD font manager to upload correct boot logo.

[image]

enterobotica, 2024-04-20:
Thanks, I came here to report the same, glad to see that I found the right solution. 

I have observed that the orientation of gyro was showing correct on Emuflight, but in Betaflight it always showed 270°CW off. The FC is mounted upside down in my 3.5" cinewhoop setup, and the it's mounted keeping the arrow facing forward. In Emuflight I only had to rotate board orientation to 180°, while in BF I have to add the angle as well.

I wanted to report the same, should I continue in this thread or create a new issue?

haslinghuis, 2024-04-20:
Have checked the emuflight config and it uses the same alignment.

@nerdCopter might know more.

Please for support consider using our [Discord server](https://discord.com/invite/gV4XSq3fus)

closed 2024-04-20 by haslinghuis

nerdCopter, 2024-04-20:
You should upload new logo or existing Betaflight font via font manager -- may need LiPo plugged.

`CW270_DEG` is defined in Emu. If it did something different, maybe a bug :laughing: . You may need to orient some "Flip" option in Betaflight.

Emu Discord if needed: https://discord.gg/BWqgBg3