betaflight/betaflight #13563: non-breaking bug -- cli pasting diffs results error after vtx settings

nerdCopter, 2024-04-21:
### Describe the bug

snippet
```

# vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917

# vtxtable powerlevels 4

# vtxtable powervalues 25 100 200 350

# vtxtable powerlabels 25 100 200 350

# 
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# # master

# set dyn_notch_count = 2
dyn_notch_count set to 2
```
```
# save
###ERROR IN save: ERRORS WERE DETECTED - PLEASE REVIEW BEFORE CONTINUING###
###ERROR IN save: PLEASE FIX ERRORS THEN 'SAVE'###

# #
```

### To Reproduce

- go to CLI, get a diff, copy to clipboard, paste that clipboard. any master for few weeks or more now (i did not bisect to find when it started).
- multiple targets.

### Expected behavior

no error on blankline or commented line 

### Support ID

```txt
Id: "277d563d-f499-4a13-8bd6-6ebcc231fd4f"
```

### Flight controller

multiple

haslinghuis, 2024-04-21:
```
# vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
###ERROR IN vtxtable: BAND NUMBER NOT BETWEEN 1 AND 0###
```

vtxtable `band` should be `bands`

haslinghuis, 2024-04-21:
This should work:

```
vtxtable bands 5
vtxtable channels 8
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 350
vtxtable powerlabels 25 100 200 350
```

nerdCopter, 2024-04-22:
- my snippet was just the area in the CLI output, not the entire vtx section.
- now it's proving me wrong :/, but i've seen another pilot post the same issue in discord, so i'm unsure the trigger.  it might be an emptyline or it might be an invisible character.
[image]

haslinghuis, 2024-05-15:
Is this still an issue?

nerdCopter, 2024-05-16:
- i've seen other pilots have the issue, but it's non-breaking and not 100% reproducible.
- might be illegitimate and/or hidden characters in the diff/dump.
- closing.

Machatron, 2024-06-14:
Please reopen. I'm consistently affected by this issue on all my quads on 4.5. here's the exact dump. pasting the vtxtable and vtx commands causes it to trigger.
[BTFL_Mach.txt](https://github.com/user-attachments/files/15830307/BTFL_Mach.txt)

haslinghuis, 2024-06-14:
@Machatron can you try https://app.betaflight.com/

Machatron, 2024-06-15:
issue persists. 
[image]

also for some reason the cli paste placed the remainder of the pid profile sets after the save command. the original dump that was loaded was not like this.
[image]

closed 2024-06-20 by haslinghuis