betaflight/betaflight #13577: DPS310 not recognized on SPEEDYBEEF7V3 using GEP-M1025-DQ

haslinghuis, 2024-04-23:
### Describe the bug

As per title. Internal baro is a BMP280. Magnetometer is detected.

Using GEP-M1025-MI barometer MS5611 and magnetometer IST8310 are detected.

### To Reproduce

Flash master and include magnetometers define.

### Expected behavior

Detect baro on external module

### Support ID

```txt
2dd928b4-b734-4a6a-bcf8-b48c5687b56b
```

### Flight controller

SPEEDYBEEF7V3

### Other components

Only SpeedyBee F7 V3 BL32 50A 30x30 Stack on lab power with mentioned module.

### How are the different components wired up (including port information)

Using connectors - as magnetometer is working there is no wiring issue.

### Add any other context about the problem that you think might be relevant here

- Tried with and without battery power.
- Setting baro_hardware or other settings does not solve the issue.

ledvinap, 2024-04-23:
there is one USE_DSP310 typo in configs, maybe this one?

ledvinap, 2024-04-29:
@haslinghuis : as @ot0tot said ... and don't expect either baro to work when both are connected

haslinghuis, 2024-04-29:
> Perhaps part of the design guidelines should include using separate I2C devices for on-board and external connections?

Agree introducing external I2C settings would address this issue.

closed 2024-05-15 by haslinghuis

Martivip, 2024-06-14:
> > Perhaps part of the design guidelines should include using separate I2C devices for on-board and external connections?
> 
> Agree introducing external I2C settings would address this issue.

HI @haslinghuis i have the same issue with my GPS unit with baro GEP-M1025-DQ and KakutteH7 FC. Magnetometer is detected but both internal FC and external baro build in GPS unit is not recognised due to described above conflict. So you are saying that after implementation external I2C settings in BF would address this issue ? any development for this is scheduled ?

haslinghuis, 2024-06-14:
Won't be a software solution.

Using I2C port 1 for internal and port 2 for external devices would already make this possible. Board would need to expose pads for second port. In this way it would not conflict with duplicate devices on I2C port 1 as it would allow to choose active port for the device to be used.

Martivip, 2024-06-14:
> Won't be a software solution.
>
> Using I2C port 1 for internal and port 2 for external devices would already make this possible. Board would need to expose pads for second port. In this way it would [...]

ok thanks for the information. Honestly there is no such a FC with second I2C port :(  ok so to use this external baro build in GPS I need to use FC without the baro or physicaly disconnect it on the FC, right ?

ot0tot, 2024-06-14:
> ok thanks for the information. Honestly there is no such a FC with second I2C port :( ok so to use this external baro build in GPS I need to use FC without the baro or physicaly disconnect it on the FC, right ?
There are many FCs that already use separate I2C busses for on-board and external sensors. 

Yes, you will need to disconnect one of the baros.

Martivip, 2024-06-14:
> There are many FCs that already use separate I2C busses for on-board and external sensors.
>
> Yes, you will need to disconnect one of the baros.

could you give me a hint how to recognise such a FC with separate busses ? should I search in FC config.h files ?

ot0tot, 2024-06-14:
> could you give me a hint how to recognise such a FC with separate busses ? should I search in FC config.h files ?

Yes, they will have multiple I2C resources defined. Example:
https://github.com/betaflight/config/blob/b98adf6f5e1acfaf4c3f9364579928cdf75a0df9/configs/JHEH743_HD/config.h#L74-L77

You will need to confirm the target resource assignments are accurate and present on the FC.

A decent manufacturer will also cover it in the manual and include it in the specs.