betaflight/betaflight #13580: GPS count of satellites are not shown in the OSD (DJI Goggle1 with WTFOS)

Quickred, 2024-04-24:
### Describe the bug

The symbol and the **count of satellites** are not shown in the OSD and also not in the DVR
Configuration in Betaflight configurator 10.10.0-RC4:
[image]
and the OSD and DVR (WTFOS overlay) looks like this:
[image]

### To Reproduce

installed BF 4.5 RC4 with video format HD. Presets "OSD for Fpv.wtf" and "SupaflyFPV Freestayle 6 Inch".
**No count of GPS satellites** are shown ion the display.  GPS satellite count is displayed in the TBS Tango2 display, also GPS rescue gives a signal for 3D FIX and **GPS RESCUE is working succesful.**

### Expected behavior

worked with BF 4.4, count of sats were shown in OSD

### Support ID

```txt
035776b9-1ec3-41fa-9749-df6860f10648
```

### Flight controller

HGLRCF722E

haslinghuis, 2024-04-24:
Try `set displayport_msp_fonts = 0,0,0,0`

Quickred, 2024-04-24:
Thank you for your fast answer !!! **IT WORKS** ! Love you all <3

Quickred, 2024-04-25:
Sorry, have not read the release notes carefully:

https://betaflight.com/docs/wiki/release/betaflight-4-5-release-notes

closed 2024-04-25 by Quickred