betaflight/betaflight #13581: CRAZYBEEF4SX1280 locks up after loading saved diff including `dshot_bidir = ON`

haslinghuis, 2024-04-24:
### Describe the bug

Loading diff locks up F411 when `dshot_bidir = ON` is present in the diff

### To Reproduce

Load saved diff to flight controller.

```
# diff all defaults

# version
# Betaflight / STM32F411 (S411) 4.5.0 Jun  5 2023 / 11:36:47 (d73092992) MSP API: 1.46
# config rev: c433987

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

# feature
feature -AIRMODE
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 30 2048 115200 57600 0 115200

# aux
aux 0 0 0 1600 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1325 1700 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 2 1550 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 20,0,56,1
set max_check = 2000
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set osd_vbat_pos = 2391
set osd_rssi_pos = 2276
set osd_link_quality_pos = 2244
set osd_rssi_dbm_pos = 2212
set osd_tim_2_pos = 2403
set osd_flymode_pos = 2371
set osd_crosshairs_pos = 2253
set osd_current_pos = 2422
set osd_craft_name_pos = 2410
set osd_core_temp_pos = 182
set osd_canvas_height = 13
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5800

profile 0

# profile 0
set i_pitch = 99
set d_pitch = 35
set f_pitch = 105
set i_roll = 96
set d_roll = 34
set f_roll = 99
set p_yaw = 48
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 0
set d_min_pitch = 0
set tpa_rate = 60

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 100
set roll_expo = 40
set pitch_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

set dshot_bidir = ON

# save configuration
save
```

### Expected behavior

Not locking up firmware.

### Support ID

```txt
N/A
```

### Flight controller

F411 related

### Other components

Tested with NOX F411, CRAZYBEEF4SX1280 and BETAFPVF4SX1280

### How are the different components wired up (including port information)

Default

### Add any other context about the problem that you think might be relevant here

Moving `set dshot_bidir = ON` to the end of the diff solves the issue

haslinghuis, 2024-04-24:
@SteveCEvans @ledvinap any idea why this could happen. Did not check flashing further back then 4.5.0 RC1.

haslinghuis, 2024-04-24:
@ledvinap @SteveCEvans any idea why this could happen? Did not check flashing further back then 4.5.0 RC1.

haslinghuis, 2024-04-25:
@ctzsnooze was experiencing this issue - so I tried to narrow it done.

@ledvinap @SteveCEvans any idea why this could happen? Did not check flashing further back then 4.5.0 RC1.

Note moving `set dshot_bidir = ON` to the end of the diff solves the issue

ledvinap, 2024-04-26:
@haslinghuis : Can you test status if pasting manually? Maybe delay before `save` is important

haslinghuis, 2024-04-26:
@ledvinap 

- It does not reach `save` 
- After moving `dshot_bidir = ON` just before save in the diff - operation is successful.
- Status after manually applying lines up to `profile 0`, then I typed `profile 0` and flight controller locks up.

```
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=61degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: UNCONFIGURED, size: 3779, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MAX7456 (30 x 13)
BUILD KEY: a53ecdf05462f8a9118ffe647edad54b (4.5.0-dev)
System Uptime: 137 seconds, Current Time: 2024-04-26T14:10:19.037+00:00
CPU:63%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - INIT)
I2C Errors: 0
GPS: NOT ENABLED
Arming disable flags: RXLOSS ANGLE LOAD CLI MSP DSHOT_TELEM

# profile 0
profile 0
```

ledvinap, 2024-04-26:
Please try moving dshot_bidir = ON just before save, but entering it manually - create delay before save

haslinghuis, 2024-04-26:
`save` seems not be the issue - as manual applying the diff still hangs firmware - operation is successful when moving that line to the end before save in the diff - without delay.

ledvinap, 2024-04-26:
So no crash when dshot_bidir = on is after profile 0 ? Even when save is delayed?

`save` reboots the board - maybe the board is rebooted before `dshot_bidir = on` can crash it. From you reply, it is not obvious if only order is important, not time.

haslinghuis, 2024-04-26:
Order is important. Not timing (as we don't reach save)

PR #13589 (2024-04-27): Use the cached value of useDshotTelemetry to ensure consistent runtime use if dshot_bidir is changed
Fixes #13581 

`motorConfig()->dev.useDshotTelemetry` was being referenced many times in the code and if the value was changed then uninitialised filters would be used etc. This PR uses the value cached upon initialisation of DSHOT, stored in `useDshotTelemetry`.
closed 2024-04-27 by blckmn