betaflight/betaflight #13582: Smooth arming when air mode is enabled

ChrisRosser, 2024-04-25:
### Is your feature request related to a problem? Please describe

When arming with airmode enabled a powerful quad that is tightly tuned can momentarily twitch/jump due to the interaction of the motors spinning up and airmode. This can make the arming process a little more scary than it needs to be!

The problem is very clear on the 3rd arm in the attached video.

https://github.com/betaflight/betaflight/assets/41840611/c27f0871-8fb6-4baa-93dd-d9af8941e3e3

### Describe the solution you'd like

I think there is a very simple solution for this issue:

1. Wait a short time (~1s) after arming before enabling air mode to allow the motors to spin up and stabilise.
2. Fade in the infludence of air mode over ~1s after the quad is armed to avoid a step change.

### Describe alternatives you've considered

No alternatives considered.

SupaflyFPV, 2024-04-25:
Hi Chris. There is an airmode_activate_throttle parameter which is default to 25 - so airmode as I understand it is not active on arming until it hits that threshold.

Can you log this behaviour?

There is also a new TPA Low functionality which is active only for arming, to help with high gain tuning/builds....

eg for a 40% reduction which switches off after the breakpoint it would be:

set tpa_low_rate = 40
set tpa_low_breakpoint = 1070
set tpa_low_always = OFF

ChrisRosser, 2024-04-26:
[Log.zip](https://github.com/betaflight/betaflight/files/15126420/Log.zip)
Hi Supafly!

Thanks for helping me with this. Here is the log file.

SupaflyFPV, 2024-04-26:
> [Log.zip](https://github.com/betaflight/betaflight/files/15126420/Log.zip) Hi Supafly!
> 
> Thanks for helping me with this. Here is the log file.

Hi Chris, no worries. The log appears to have an unprovoked gyro output on arming, and because he is using the Speedybee v3 stack I suspect that it is one of the many I've seen which has issues due to a sub-optimal gyro voltage regulator circuit. The issue tends to appear as sporadic extreme pulses of gyro data way off anything commanded or from disturbance.

Thankfully they seem to have solved it with the v4.

xviews, 2024-04-28:
How can I contact Brandon, I also have the problem and a v3

closed 2024-06-05 by github-actions[bot]