betaflight/betaflight #13586: Blackbox altitude units for GPS-track

HereticToXa, 2024-04-26:
### Is your feature request related to a problem? Please describe

After decoding blackbox log from Betaflight the gpx file contain information about altitude in wrong units.

### Describe the solution you'd like

It seems like it should be multiplied by 0,1.
So the scale is just off by one decimal.

### Describe alternatives you've considered

CLI command for choosing the right units.

### Other information

There were some requests for it, but it was closed somehow. And I heard INav have no such a problem.

jannikbecher, 2024-05-27:
Hey @HereticToXa :)

The altitude is saved as an integer with 0.01m resolution
https://github.com/betaflight/betaflight/blob/6ba4944a9fd5d92da9c2ebfa1d983a2eca391e1b/src/main/io/gps.h#L224-L229

and written to the blackbox with 0.1m resolution
https://github.com/betaflight/betaflight/blob/6ba4944a9fd5d92da9c2ebfa1d983a2eca391e1b/src/main/blackbox/blackbox.c#L1113

so this is wanted behaviour.

jannikbecher, 2024-05-27:
I submitted a [PR](https://github.com/betaflight/blackbox-log-viewer/pull/748) in the log viewer to fix the GPX output

HereticToXa, 2024-05-27:
Thanks, but in BF 4.5 this "bug" was fixed i think. Now export gpx from blackbox explorer works great

jannikbecher, 2024-05-28:
Ok perfect, so the issue can be closed? :)

HereticToXa, 2024-05-29:
Well. I'm not sure, because not everybody will update to betaflight 4.5 to fix this issue. So I think it must be fixed anyway

haslinghuis, 2024-05-29:
Both snippets above were there at least since firmware 4.0

closed 2024-05-29 by haslinghuis

HereticToXa, 2024-05-30:
O, yeah? And why on 4.3 many people got this error in vertical scale after exporting gpx from blackbox logs? 🤔