betaflight/betaflight #13590: Dashboard build option impedes Betaflight initialization and creates I2C errors.

B1naryShad0w, 2024-04-27:
### Describe the bug

When adding "Dashboard" as a build option to Betaflight, certain I2C conditions make the firmware take an entire 30 seconds to initialize. This issue is specific to Betaflight 4.5 RC4, and does not exist in 4.4.3. I have not tested other 4.5 release candidates. My testing is based around the I2C errors readout in the configurator, as it seems to have some correlation with respect to the I2C errors from my baseline testing in 4.4.3.

### My hardware:

- [Kakute F7](https://cdn.shopify.com/s/files/1/0604/5905/7341/files/Holybro_Kakute_F7_Manual.pdf) which has a built in I2C barometer (which I suspect is connected to the same hardware I2C interface as the SDA/SCL solder pads on the board based on the observations that follow).
- BN-880 GPS (with built in I2C magnetometer).

### Other considerations:

- I have no other I2C devices connected.
- My BN-880 (and thus, my compass) is powered by my ESC. It remains unpowered with just the USB connected and is only powered when supplied through the XT60 on my drone.
- The TX/RX from my BN-880 is connected to UART3. I have not been able to configure and get the GPS running yet, but I doubt it is a related issue. The magnetometer works as expected with 4.4.3 and even in 4.5RC4 so I assume that the GPS does not interfere with these tests.

### Notes:

- "**Disconnected**" means I powered the FC via USB with the BN-880 disconnected from it's connector and is not connected to the I2C pads.
- "**Connected Unpowered**" means I powered the FC via USB with the BN-880 connected to the I2C pads, but did not power the device via the XT60.
- "**Connected Powered**" means I powered by whole drone via XT60 wile the compass is connected to the I2C pads, **and then** plugged in the FC via USB shortly after.
- In all cases where the compass is connected but unpowered the barometer is undetected. I suspect this is hardware related but I have not found any continuity between the SDA and SCL pins with respect to ground (nor the 5v) on the BN-880 that would indicate that it's holding signals low. I have not had the chance to test this with an oscilloscope and assume this is not a supported configuration for the BN-880 but nonetheless included my findings below.
- All tests with 4.5RC4 have the magnetometer enabled, even when disconnected.
- I tried my best to be precise with the exact error count (e.g. between 1027 and 1028) as I've found them to differ between cases.
- Full power cycle after flashing each firmware.

Some test cases have errors incrementing by 1 every 5 seconds. This is noted.

Some test cases involve Betaflight not initializing for 30 seconds. This is determined by the blue status LED not flashing and being unable to connect through the configurator although the COM port is visible and detected immediately after plugging in. After 30 seconds the LED starts flashing and I am able to connect. If the BN-880 is powered via XT60 **after the flight controller is plugged into USB but before the 30 seconds** while in this state, Betaflight still takes the full 30 seconds to initialize from initial power-on via the USB.

**4.4.3 with both Core+Magnetometer and Core+Magnetometer+Dashboard:**
Disconnected: 7 errors (0 if magnetometer is disabled in configuration.)
Connected Unpowered: 8 errors (1 if magnetometer is disabled.)
Connected Powered: 0 errors

**4.5RC4 with Core only:**
Disconnected: 0 errors
Connected Unpowered: 1 error
Connected Powered: 0 errors

**4.5RC4 with Core+Magnetometer**:
Disconnected: 9 errors
Connected Unpowered: 10 error
Connected Powered: 0 errors

Here is where behavior is abnormal:

**4.5RC4 with Core+Dashboard:**
Disconnected: 1027 errors with increment
Connected Unpowered: Long initialization and 1028 errors with increment
Connected Powered: 1028 errors with increment

**4.5RC4 with Core+Dashboard+Magnetometer:**
Disconnected: 1036 errors with increment
Connected Unpowered: Long initialization and 1037 errors with increment
Connected Powered: 1027 errors with increment

I would love to help get to the bottom of this so if you need me to test some updated firmware, install some verbose firmware for more detailed errors, or if you guys need any more information to get to the bottom of this please let me know.

### To Reproduce

Simply install the affected firmware leaving default settings, with peripherals connected as described.

### Expected behavior

Behavior similar to 4.4.3.

### Support ID

```txt
53e97ffa-3ebc-458a-a27e-d90329481095
```

### Flight controller

Kakute F7

ledvinap, 2024-04-29:
Just take approach suggested by @hydra and don't enable features just because they are compiled in.

Also, dashboard should throttle reinitialization attempts

B1naryShad0w, 2024-04-30:
> Why are you including the DASHBOARD define?
> 
> @haslinghuis Should we just remove all `#define DASHBOARD_I2C_INSTANCE` lines from configs since this feature is almost never used?

I saw in [the documentation](https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#other-options) that it "enables the post-flight overview screen," which led me to assume it to mean the post flight statistics found in the OSD tab. I see that I was mistaken and that the post flight statistics doesn't seem to be related to the DASHBOARD build option.

I am curious as to what this define does exactly and when it should be used as I've always had it enabled under the wrong assumption before 4.5, but don't notice any missing features without it.

haslinghuis, 2024-04-30:
It's for an external [O]LED display. See https://betaflight.com/docs/development/display

ctzsnooze, 2024-05-04:
Updated the user documentation in [PR 418](https://github.com/betaflight/betaflight.com/pull/418)

PR #13608 (2024-05-21): Do not enable features by default
- Fixes #13590
- Fixes #13602
- See also #13494 which should be merged first
- Remember #13244
closed 2024-05-21 by haslinghuis