betaflight/betaflight #13591: 044b1e0d-2d0b-4439-b8d8-28769b8ea1c4

vichevg-gDEV, 2024-04-27:
### Is your feature request related to a problem? Please describe

The latest version of betaflight does not recognize the gyro and keeps prompting me a message saying there's no motor output protocol selected, so I can't really configure it

### Describe the solution you'd like

Could you please check what I can do about it and write somewhat of a guide

### Describe alternatives you've considered

Or maybe just point me to a vid that would be helpful so I can setup the drone

ot0tot, 2024-04-29:
>  board: manufacturer_id: STMI, board_name: STM32F4DISCOVERY

You have the wrong Betaflight target flashed.  What FC do you have?

haslinghuis, 2024-05-12:
Discord for issues, github for development.

closed 2024-05-12 by haslinghuis