betaflight/betaflight #13598: Motors on F72x not working with final 4.5

TheIsotopes, 2024-04-28:
### Describe the bug

since final release the motors on my f7 quad not working anymore.

### To Reproduce

using latest final on MATEKF722MINI

plug in the battery, arming the quad and use the throttle but the motors dont turn

motor beeper also not working

### Expected behavior

running motors on final release

### Support ID

```txt
Id: "f69576ff-9b3e-41fb-ae2b-c25f7b783118"
```

### Flight controller

MATEKF722MINI

hendraramadani, 2024-04-28:
same problem 
[image]

SteveCEvans, 2024-04-28:
Make sure your ESC is up to date. I’ve just been flying an F722 on 4.5.0 without issue.

TheIsotopes, 2024-04-28:
@SteveCEvans I'm using latest BlHeli_32 32.10. I haven't seen anything more recent yet.

haslinghuis, 2024-04-28:
Recommended firmware for [BLH32](https://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware#blheli_32)

TheIsotopes, 2024-04-28:
@haslinghuis 
OK, that's new to me. I'll test it right away.
Have these problems with Blheli_32 already been discussed with the developer?

haslinghuis, 2024-04-28:
@TheIsotopes 
We are moving in direction of BlueJay and AM32.

One report with reporting an issue here: `https://github.com/bitdump/BLHeli/issues/713`

TheIsotopes, 2024-04-28:
@haslinghuis ok, the same problem also occurs with blheli_32 version 32.7.

Any other good advice?

TheIsotopes, 2024-04-28:
@haslinghuis @SteveCEvans 
For your information: I don't use any rpm filter or telemetry, so no `dshot_bidir` either.
I couldn't find the problem on my f405 quads, everything works as expected here.

haslinghuis, 2024-04-28:
- Just tested AOCODARCF405 with TBS69 with bidirectional Dshot enabled. No issues except beeper not working.

- Nothing merged after https://github.com/betaflight/betaflight/commit/8d1d8872d6901fb834adbaa5181302ff9ace82f9 which would explain this behavior. 

- I know we are struggling with `FAST CODE` on STM32F7X2 but this does not look alarming either:

`ITCM_RAM:       14512 B        16 KB     88.57%`

- Might try to disable bitbang:
`set dshot_bitbang = AUTO` => `OFF`

TheIsotopes, 2024-04-28:
@haslinghuis ok with `set dshot_bitbang = OFF` beeper and motors working again.

Should the newer versions of blheli_32 generally not be flashed or do the errors only occur with activated rpm filtering?

haslinghuis, 2024-04-28:
@TheIsotopes 

- Just tested AOCODARCF405 with TBS69 with bidirectional Dshot enabled. No issues except beeper not working.

- Nothing merged after https://github.com/betaflight/betaflight/commit/8d1d8872d6901fb834adbaa5181302ff9ace82f9 which would explain this behavior. 

- I know we are struggling with `FAST CODE` on STM32F7X2 but this does not look alarming either:

`ITCM_RAM:       14512 B        16 KB     88.57%`

- Please try 4.6.0-dev build:

[betaflight_4.6.0_STM32F7X2_MATEKF722MINI.zip](https://github.com/betaflight/betaflight/files/15142936/betaflight_4.6.0_STM32F7X2_MATEKF722MINI.zip)

TheIsotopes, 2024-04-28:
your firmware version only works also with `set dshot_bitbang = OFF` by using 32.8 and 32.10

haslinghuis, 2024-04-28:
If you could test RPM filter is working with dshot_bitbang disabled we could update board configuration for MATEKF722MINI.

TheIsotopes, 2024-04-28:
With your version from above and `set dshot_bitbang = OFF` the  RPM filter will deactivated.

haslinghuis, 2024-04-28:
@SteveCEvans 

Is there a way to improve `dshot_bitbang = AUTO` detection depending on `dshot_bidir` setting?

pkojder, 2024-05-06:
Same issue with SpeedyBee F7 V3, motors not spining at all (tested in BF Configurator, BlheliSuite32 and after arming a quad from radio). ESC are up to date (32.10). Moreover, enabling Bidirectional DSHOOT shows rpmfilter flag and Poblems with comunication occurs via USB and Bluetooth. After downgrading to 4.3.3 works as a charm. 
BF 4.5 is definatelly realised to soon. Guys you have screwed.

spadeadam, 2024-05-07:
Same issue with Airbot F7 and Blheli-S 4 in 1 Esc, everything works with BF 4.4 then “upgraded” to BF 4.5.0 No motor output , tried DShot 300 and DShot600. Flashed back to BF 4.4 everything works OK again ?

kostamilorava, 2024-05-08:
Enabling `dshot_bidir` fixes this problem for `Matek F722 MK4`  [Link here](https://betaflight.com/docs/wiki/guides/current/dshot-rpm-filtering#enabling-bidirectional-dshot)

blckmn, 2024-05-12:
> Same issue with SpeedyBee F7 V3, motors not spining at all (tested in BF Configurator, BlheliSuite32 and after arming a quad from radio). ESC are up to date (32.10). Moreover, enabling Bidirectional [...]

Did you try setting dshot bit bang = off?

SteveCEvans, 2024-05-12:
To confirm, is this issue only being seen with bidirectional DSHOT disabled?

haslinghuis, 2024-05-12:
In my understanding 
- bitshot_bitbang should be enabled for dshot telemetry
- when dshot_bidir is disabled bitbang should be disabled. ????

SteveCEvans, 2024-05-12:
Just checked DSHOT600 with a Holybro Tekko32 F3 Metal ESC and TMOTORF7 FC.

With bi-drectional DSHOT disabled I see:

[image]

And with it enabled I see:

[image]

So, we're getting the expected inversion, telemetry is working, and the error count I'm seeing as zero.

This FC also has the motor resources spread over 2 ports, so multiple ports can't be an issue.

```
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 B00
resource MOTOR 4 B01
```

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_SDI 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_SDO 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_SDI 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: SPI_SDO 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 2
```

SteveCEvans, 2024-05-12:
@TheIsotopes Please test https://github.com/betaflight/betaflight/pull/13633

closed 2024-05-12 by blckmn

TheIsotopes, 2024-05-13:
@SteveCEvans working again, thanks for fixing it.

kojderek, 2024-05-30:
> @SteveCEvans working again, thanks for fixing it.

How was it fixed. What shoud I do to run my motors with BF 4.5.0

DeviousPenguin, 2024-06-02:
> Did you try setting dshot bit bang = off?

Hi, I'm also running the SpeedyBee F7 V3 with the SpeedyBee BL32 50A 4-in-1 ESC, and I seem to also be getting this issue also.

My motors were working great with BF 4.3 that was pre-installed on the controller, and I've backed up my config also but it was pretty much default config.

After updating to BF 4.5, I can no longer spin up the motors in the configurator, even the test motor tab doesn't spin up the motors.

I'll give 'dshot bit bang = off' a try and see if this helps at all

haslinghuis, 2024-06-02:
Just try 4.5.1

AlexRiQ, 2024-06-09:
Hello, im here with the same problem as anyone else. Speedybee esc BL32, latest version. Motors dont work with bidirectional dshot even in 4.5.1

TheIsotopes, 2024-06-22:
Since BlHeli_32 is now being laid to rest, i am currently testing the AM32 firmware on one of my quads. To be more precise, i am flying the AM32 V2.12 on Aikon AK32 35A with an F72x FC and activated bidir Dshot. So far I have not noticed any problems. @blckmn @haslinghuis @SteveCEvans What about you?

On [AM32](https://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware#am32) only versions 2.05 and older are listed. That's why i wanted to ask here.

haslinghuis, 2024-06-23:
See 3.03 available [here](https://github.com/AlkaMotors/Am32-Experimental/releases) :smiley: 

Also 2.01 is last public [available](https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware/releases)

Documentation is always lagging.

@AlkaMotors any info on 2.12 used here?

TheIsotopes, 2024-06-23:
Quote from old repo : 
> All mcu's have been combined to one project. Please use the new repository [here](https://github.com/am32-firmware/AM32).

@haslinghuis AM32 have a new repo where you can find the 2.12 version.

spadeadam, 2024-06-23:
Has anyone worked out what changed in the “upgraded” firmware 4.5.0 to stop the motors working with F7xx CPU’s ?
Esc is 4 in 1 Blheli-S with latest firmware….no problems with previous BF firmwares
Currently staying with ver. 4.4

haslinghuis, 2024-06-23:
@spadeadam - configuration issue is solved in 4.5.1

kojderek, 2024-06-28:
> @spadeadam - configuration issue is solved in 4.5.1

What configuration issue. On some FCs it does not work at all, despite the setting. Why are You affraid to confes that 4.5 is bugged.

haslinghuis, 2024-06-28:
@kojderek provide support ID and information about hardware [revision], setup, firmware version and settings used where it does not work to enable investigation.